VHDL-LS/rust_hdl #210: Avoid sending diagnostics for all files on every change

kraigher, 2023-11-05:
The LSP protocol supports client caching for diagnostics on a file-level. Currently VHDL-LS publishes all diagnostics for every edit. This works well when there are few diagnostics but is not optimal. When adding dead code detection many code bases could have larger amount of diagnostics. We could thus optimize this. 

Another benefit is that it simplifies the code inside vhdl lang since caching does not need to be handled there.

lgu-appear, 2024-04-03:
Just tagging @Schottkyc137 here

Schottkyc137, 2024-04-09:
> Just tagging @Schottkyc137 here

By tagging I assume you want this prioritised?

lgu-appear, 2024-05-13:
> By tagging I assume you want this prioritised?

That would be very nice indeed!

Schottkyc137, 2024-06-09:
@kraigher just to confirm, I couldn't find any information about the client actually being _forced_ to to cache diagnostics. However, there is the [pull-based diagnostic model](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_pullDiagnostics) where the client periodically asks the server to compute diagnostics for a certain file. There, one can send an [UnchangedDocumentDiagnosticReport](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#unchangedDocumentDiagnosticReport), indicating that a file has no new diagnostic, and the old version should be used. Is this what you are referring to? If so, are you suggesting to switch to this new model or keeping both, using the push-based version when the pull-based is not available?

kraigher, 2024-06-10:
From the LSP spec:

> When a file changes it is the server’s responsibility to re-compute diagnostics and push them to the client. If the computed set is empty it has to push the empty array to clear former diagnostics. Newly pushed diagnostics always replace previously pushed diagnostics. There is no merging that happens on the client side.

This means the client has to remember what diagnostics the server sent until the server sends another publish diagnostics for the file. 

I do not think it is a good model for the client to poll the server for diagnostics. That is only useful for simple LSP servers where checks are only done in a single file. In a project wide LSP a removed definition in one file would trigger an unknown variable error in another file. You always wants the full diagnostics of the entire project tp be updated and only the server can manage that.

Schottkyc137, 2024-06-10:
Ah, I see what you mean. Makes sense to me, even though the pull-based approach also has methods to resolve the inter-file dependency. Still, I think the push-based approach is better.
I'm a bit confused, however, about the "it simplifies the code inside vhdl lang since caching does not need to be handled there" part. How would you optimise caching? Not sending diagnostics for every file means that we need to check whether we need to re-send diagnostics for every file after analysis. Without pretty big adaptations to the analysis stage and / or more complex dependency tracking, I don't see how this would work?
Or are you only referring to the parser diagnostics, since parser diagnostics can only affect a single file?

kraigher, 2024-06-10:
The main use case for this issue was someone having 10000 warnings that became slow to resend over json rpc for every change. If there as a cache in the vhdl-ls crate that knew the diagnostics it published for every file before it could avoid resending every diagnosic to the client on every change and only publish diagnostics for files that changed.

PR #310 (2024-07-01): Avoid sending all diagnostics on every file change
Closes #210

This PR implements a server-side diagnostic cache, containing analysed diagnostics. On every change, new diagnostics are evaluated on the server side. Only when the diagnsotics are not in the cache are they then sent to the client.
This means that there are typically less transactions between server and client.

I'm not quite sure whether this resolves some of the performance impact seen by sending too many diagnostics as the server now has to do a lot of comparisons to check whether the diagnostics are in the cache.
closed 2024-07-01 by Schottkyc137