VHDL-LS/rust_hdl #212: Autocompletion / Intellisense not working for function arguments (VSCode)

AndrzejKowalski9917, 2023-11-07:
While typing in VSCode, it normally suggests autocompletion. Starting with VHDL-LS 0.67.0 triggering this autocompletions for function arguments fails in a larger project I'm working at.  Outside of function arguments, the feature works as intented.

I activated tracing messages in the server and it seems, that getting a response in this case takes more than 3 seconds which leads to the autocompletion feature failing. With VHDL-LS 0.66.0 the same response from the server only took 1 ms.

Here you see the behavior with VHDL-LS 0.66.0. Suggestions for autocompletion pop up immediately.
[image]

This is the behavior with VHDL-LS 0.67.0 (also with all following up to 0.71.0). After hitting CTRL+SPACE to trigger the autocompletion manually, it seems to time out and fails:
[image]

Unfortanately this is not reproducible with a minimal example small project and I can't share my project. If I should test something else or can help otherwise, please let me know.

kraigher, 2023-11-07:
@Schottkyc137  I think this is related to the new autocompletion additions. It seems there is something very inefficient about them that starts to show when there is a large project. The fact that it takes 3 seconds tells me its just inefficient but not an infinite loop. I think we should disable the functionality on master until this is investigated.

kraigher, 2023-11-07:
@Schottkyc137 I think a problem is that the AutocompletionVisitor will traverse the entire AST of all files instead of early exiting. Also because the Visitor uses boxed data it becomes very inefficient since millions of ASTNodes are allocated on the heap. I think if the AutocompletionVisitor could avoid traversing the entire AST of the project this inefficency would not be so noticable.

kraigher, 2023-11-07:
@AndrzejKowalski9917 I just released 0.72.0 that removes the part of autocompletion that I suspect is causing the problem. Could you confirm?

Schottkyc137, 2023-11-07:
I also suspect that the new completion implementation is causing the lags. I didn't consider large projects (also because all of the examples I'm currently working with are only moderate in size). The same is true for example project (which is fairly large already). This didn't cause the issue as well. I'll fix it as soon as I can but at the moment I'm somewhat busy (shouldn't be too tricky to fix though). I think I have some time tomorrow to work on this.

kraigher, 2023-11-07:
@Schottkyc137 I disabled it on master to reduce the stress of fixing it.

AndrzejKowalski9917, 2023-11-07:
@kraigher I just tested it with version 0.72.0 and can confirm that it is working again as well as it was in version 0.66.0.

kraigher, 2023-11-07:
Great, thanks for confirmation

kraigher, 2023-11-07:
We can keep the issue open until we fix the feature we temporarily disabled

PR #215 (2023-11-08): Fix visitor is slow for large designs
Closes #212. The `walk`  methods now only operate on a single file, and not the entire AST.

The issue can actually be checked with the example project. However, one must trigger completions inside a `MapAspect`. With this solution, completing takes a negligible amount of time (< 5 ms) compared to the pre 0.7.2 release version (~700ms).

I noticed another issue here, namely that some completions are not shown for some entities. Mostly this happens when the entity is not in the same design file, but not always. I think this has something to do with referenc [... cut]
closed 2023-11-08 by kraigher