VHDL-LS/rust_hdl #217: Avoid reporting unused function declarations when constructing LUT

Nabav, 2023-11-09:
Sometimes functions are used without arguments to construct a look-up-table array (at compile time) like example [below](https://stackoverflow.com/a/70322215).

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity main is
...
end main;

architecture Behavioral of main is

  type w32_array is array(0 to 15) of std_ulogic_vector(31 downto 0);

  function powers_of_ten return w32_array is
    --     ^^^^^^^^^^^^^ <- Warning here: Unused declaration of function powers_of_ten
    variable res: w32_array := (others => (others => '0'));
  begin
    for i in 0 to 9 loop
      res(i) := to_stdulogicvector(10**i, 32);
    end loop;
    return res;
  end function powers_of_ten;

  constant pw10: w32_array := powers_of_ten;

begin

  output <= pw10(to_integer(input));

end architecture Behavioral;
```

The function is called for initialization of a constant without any parentheses.

The function declaration gets the warning

```
Unused declaration of function powers_of_ten
```

although it has been used.

kraigher, 2023-11-09:
Have you tried your own exact reproducer? It does not produce the same result for me.

Nabav, 2023-11-24:
You are right. It works perfectly with LUT construction. Sorry for the wrong example.

The problem is when the return type is a record.

I close this and follow the issue here: https://github.com/VHDL-LS/rust_hdl/issues/229.

closed 2023-11-24 by Nabav