VHDL-LS/rust_hdl #225: Missing syntactical elements

Schottkyc137, 2023-11-18:
The following is a list of elements that are currently not syntactically analyzed. Mostly this is because these elements are rarely used. However, for full compliance with the standard, these should be analyzed eventually.
Some of these can also be a starting point for new developers as these features (e.g., groups) are usually small and self-contained.
Below are also minimal reproducible examples that do not parse as of today.

This is also a good place to request prioritization on a feature if this feature is needed in a codebase.
- [ ] Package declaration inside a declarative part
  ```vhdl
  entity ent is
  end entity;

  architecture arch of ent is
    package my_pkg is
        -- ...
     end package;
  begin
  end arch;
  ```
- [ ] Package body inside a declarative part
  ```vhdl
  entity ent is
  end entity;

  architecture arch of ent is
    package my_pkg is
        -- ...
     end my_pkg;
    package body my_pkg is
        -- ...
     end package body;
  begin
  end arch;
  ```
- [ ] Disconnect Specification
  ```vhdl
  disconnect sig after 100 ns;
  ```
- [ ] Missing keywords and context where they are used
  - [ ] Sequence
  - [ ] Property 
- [ ] Group declaration
  ```vhdl
  group my_group : template_name ( A, B ) ;
  ```
- [ ] Group template declaration
  ```vhdl
  group template_name is ( sequence, function ) ;
  ```
- [ ] Concurrent select guarded and force assignment
   ```vhdl
   with a select b <= force c when d, e when f;
   with a select b <= guarded c when d, e when f;
   ```
- [ ] Question mark after select
   ```vhdl
   with a select? b <= c when d, e when f;
   ```
- [ ] Release assignment
   ```vhdl
   a <= release out;
   ```

Chris44442, 2024-05-30:
Hi, I haven't worked with the code base before. I have tried fixing the first of your issues [here](https://github.com/Chris44442/rust_hdl/tree/add_declarative_package_declaration)

I didn't understand what to do in some parts, e.g. in the analyze_declaration function or the HasEntityId impl. I have marked it with TODO. Either way your example code parses now and the package is registered as a LRM 4.7 Package declaration, instead of a LRM 4.9 Package instatiation declaration.

Should I make a PR?

Schottkyc137, 2024-05-30:
> Hi, I haven't worked with the code base before. I have tried fixing the first of your issues [here](https://github.com/Chris44442/rust_hdl/tree/add_declarative_package_declaration)
>
> I didn't [...]

Hi,
first of all thanks a lot for your interest in contributing!
What I have seen looks good so far. I think, that you should indeed open a Draft PR and then I can comment on the bits and pieces that are still unclear.

Chris44442, 2024-05-30:
Done.

I will try to add your second task (the package body) too, since I think it should be more or less a similar approach.