VHDL-LS/rust_hdl #229: Avoid reporting unused function declarations when constructing constant records

Nabav, 2023-11-24:
Sometimes functions are used without arguments to construct a constant record (at compile time) like example below.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity main is
  port (
    output1 : out natural;
    output2 : out natural
  );
end main;

architecture behavioral of main is

  type my_record_t is record
    field1 : natural;
    field2 : natural;
  end record my_record_t;

  function my_function return my_record_t is
    --     ^^^^^^^^^^^ <- Warning here: Unused declaration of function my_function
    variable res : my_record_t := (others => 0);
  begin
    res.field1 := 1;
    res.field2 := 2;
    return res;
  end function my_function;

  constant CONST1 : natural := my_function.field1;
  constant CONST2 : natural := my_function.field2;

begin

  output1 <= CONST1;
  output2 <= CONST2;

end architecture behavioral;
```

The function is called for initialization of a constant without any parentheses accessed by its fields.

The function declaration gets the warning

```
Unused declaration of function my_function
```

although it has been used.

kraigher, 2023-11-24:
I can reproduce the problem. The unused declaration is just the symptom. The problem is the analysis does not set the reference on then `my_function` part of `my_function.field1`. You hover over the `my_function` and nothing shows up like it should. Because of this information is used by unused declarations it also shows up there. I think this should be a quick fix.

closed 2023-11-24 by kraigher

kraigher, 2023-11-24:
Fixed and released now in 0.77.0

commit 6ddd5e7d (2023-11-24): Fixes reference not set for selected function call without arguments. Closes #229