VHDL-LS/rust_hdl #233: Create a dependency graph of a big project

javValverde, 2023-11-30:
Hi,
I am interested in making a dependency graph of a big project, and I think this tool is a great match for it.
Ideally I interface via python with the VHDL language server.
Can you please provide me with some infos as how can I do this. Is there any documentation/examples where I can check the API of the server?

kraigher, 2023-11-30:
The project is organized as two crates:
- The `vhdl_lang` library crate is the core part that parses and analyses VHDL and provides all the languages queries.
- `vhdl_ls` executable crate implements the language server protocol using `vhdl_lang` under the hood.

The dependency graph between all design units is created internally in `vhdl_lang`. You cannot access it via the public API at the moment.  We are currently disucssion how a public API should look like #219. It would be reasonable to have a way to query the dependencies of design units using this API and I think it should be added there.  We will see when we get to it though, as this is just an unpaid hobby project I will not make any promises on any timeframes.

What I can say for sure is I have no interest in adding any Python wrappers to `vhdl_lang` in this repository. That you would have to create yourself in your own repo.

javValverde, 2023-12-01:
It actually sounds great and I look forward to it.
I have no problem writing the python bindings myself. Is the interfacing rpc/json based? Or rather calling rust from python?

Thanks for the reply and for the great project!

javValverde, 2023-12-01:
And of course, let me know if there is any draft/beta API that I can play with