VHDL-LS/rust_hdl #239: Attribute specification must be in the immediate declarative

Nabav, 2023-12-20:
I get the message `Attribute specification must be in the immediate declarative part` with severity of 8 for any attribute of Xilinx interfaces copied from Xilinx Vivado templates.

Please use the code below to regenerate. Thanks.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity gpio_emio is
  generic (
    GPIO_WIDTH : natural := 1
  );
  port (
    gpio_ps_i   : out std_logic_vector(GPIO_WIDTH - 1 downto 0) := (others => '0');
    gpio_ps_o   : in std_logic_vector(GPIO_WIDTH - 1 downto 0);
    gpio_ps_t   : in std_logic_vector(GPIO_WIDTH - 1 downto 0);
    some_input  : in std_logic;
    some_output : out std_logic
  );
end gpio_emio;

architecture rtl of gpio_emio is
  attribute X_INTERFACE_INFO              : string;
  attribute X_INTERFACE_INFO of gpio_ps_t : signal is "xilinx.com:interface:gpio:1.0 PS_GPIO0 TRI_T";
  attribute X_INTERFACE_INFO of gpio_ps_o : signal is "xilinx.com:interface:gpio:1.0 PS_GPIO0 TRI_O";
  attribute X_INTERFACE_INFO of gpio_ps_i : signal is "xilinx.com:interface:gpio:1.0 PS_GPIO0 TRI_I";
begin
  gpio_ps_i(0) <= some_input;
  some_output  <= gpio_ps_o(1);
end rtl;
```

kraigher, 2023-12-20:
Move the attributes into the entity declaration after the port list. This is a grey area of the standard, lets say you have several architectures for the same entity, which attribute should apply to the port? Many tools such as Xilinx have allowed this pattern anyway, maybe out of ignorance that the entity can also define attributes inside its declarative part.

I am not against relaxing this rule in VHDL LS but it would take some work and consideration.

Schottkyc137, 2023-12-22:
I recently worked with Xilinx Attributes as well and (at least in 2020.2) Vivado does not seem to recognize attributes if they are placed in the immediate declarative part of an entity. You can only put them into the declarative part of an architecture. I did not check what Vivado does when you use multiple architectures for and entity thou.

kraigher, 2023-12-22:
One improvement in VHDL LS could be ignoring the error in this situation but not setting the attribute internally so foo'attr would not still not work. Might be ok for synthesis directives. 

For foo'attr to work the entity and architecture would need to be atomically analyzed so that the port attributes could be safely mutated before analyzing any other design units that could observe the port. This is a more complex change.

Schottkyc137, 2024-04-07:
I'll close this as you can now ignore the error as follows in the `vhdl_ls.toml`

```toml
[lint]
misplaced_attribute_spec = false
```

Maybe there is a better alternative in the future to allow attribute specifications in the declarative part of an entity, but for now this is the go-to solution. If a more flexible solution is required, I urge anyone to open a new issue or re-open this issue.

closed 2024-04-07 by Schottkyc137