VHDL-LS/rust_hdl #255: [FEATURE] Add exclusion list for problems panel

nohrel, 2024-01-30:
**Feature Description** :

Let user add a list of files for which no errors/warnings should be raised.

**Feature Usecase** :

Defining libraries in "vhdl_ls.toml" which raise tons of errors due to false errors seen by VHDL_LS.
- Those libraries need to be parsed for project, but they can be ignored for syntax errors 

Examples of errors that can currently be triggered : 
- Functions overriding with "std_logic_vector" and "std_ulogic_vector" -> "Duplicate declaration of "FunctionName" with signature "std_logic_vector"
_Both types are detected as identical

- Encrypted library files -> "Wrong Token" 

- etc...

befedo, 2024-01-30:
Hi @nohrel 

Isn't `is_third_party` the switch you're looking for?

from the Readme:
> Libraries can be marked as third-party to disable some analysis warnings, such as unused declarations

nohrel, 2024-01-31:
Hi @befedo 

Unfortunately, the `is_third_party` switch does not solve this issue :

- It still shows "third-party" libraries errors (seems to only disable "some warnings")
- I would also like to "exclude" a subset of files from the "work" library (without disabling analysis for all files in work library)

Examples of errors raised using the `is_third_party` switch, on ModelSim libraries (modelsim_lib, std_developerskit)

[image]

kraigher, 2024-01-31:
is_third_party only disables the unused code detection, not all errors.

You almost never want unused code warnings from third party code. Errors however you need even for third party code. The third party code still needs to be understood by the tool to check your use of it.

nohrel, 2024-01-31:
My understanding is that, when VHDL_LS identifies errors in a `.vhd` file : 
- When the file is a "work library" module (entity/arch), VHDL_LS will raise errors in **all** other `.vhd` files instantiating that module 
- When the file is a "work library" package, or when the file is from another library, VHDL_LS will not raise errors for other `.vhd` files using the package or library

It seems that "excluding" some `work` packages, or `third party libraries`, from the Error Report Panel would not have an impact on VHDL_LS capability to correctly parse the main code that uses them 
(example : constants, functions, procedures, components are still recognized in files with errors).

Regarding the external tool chain (simulation, synthesis), it seems that few of their own VHDL libraries include some VHDL syntax errors seen by VHDL_LS without impacting their tool. 

**Small clarification on the purpose of a "exclusion list" in my point of view**
Being able to monitor, in a project, only "real" impactful errors in the VSCode "Problems" panel when using VHDL_LS, by masking all "known" issues such as wrong VHDL syntax from third party libraries or work files (such as encrypted files), that we know won't impact simulation or synthesis tools, and won't impact VHDL_LS parsing capability (recognition of constants/functions/procedures/components...)