VHDL-LS/rust_hdl #256: [BUG] Function override error with types "std_logic_vector" and "std_ulogic_vector"

nohrel, 2024-01-30:
**Bug Description:**

When a function is declared 2 times with 2 different signatures, but the only difference is the use of "std_logic_vector" and "std_ulogic_vector", a <duplicate error> is raised : 
_"Duplicate declaration of 'FunctionName' with signature [STD_LOGIC_VECTOR,...]"_

**Screenshot**
[image]

[image]

kraigher, 2024-01-30:
VHDL function/procedure signature uniquenessdoes not consider subtypes, only base types. For example you cannot have two different functions where one takes natural and the other integer. std_logic is a subtype of the std_ulogic base type so they both have the same base.

nohrel, 2024-01-31:
Thank you for your quick response @kraigher 

I understand. I'll close my issue then.

I was mislead by seeing few libraries showing up errors due to this kind of overloading with same base type (some libraries from ModelSim).

closed 2024-01-31 by nohrel