VHDL-LS/rust_hdl #261: Enhancement: Improve cyclic dependency check to allow recursive instantiation

Derisis13, 2024-02-18:
Hi,

I've started using this language server recently, and it's incredible how sophisticated it is compared to all I've previously tried.
However running the LSP on the codebase I work on, it has come to my attention, that the cyclic dependency resolver is way stricter than what's possible without infinite recursion.

One of my IP cores uses recursive self-instantiation, similar to the code in this stack overflow issue: https://stackoverflow.com/questions/38567544/recursive-self-instantiation-component-vhdl (except he uses configurations and I use `if-else generate` statements combined with entity instantiations)
I can simulate it with Modelsim and even synthesize it with Quartus, but rust_hdl flags it as an error. (and stops evaluating the instantiated IP)
As far as I know self-instantiation is not against the vhdl standard, so it should be allowed by rust_hdl too.

I don't want to ask the impossible: a quick read on the subject has made it clear that this is an example of the halting problem, and it's impossible to solve. I don't have a good idea of how to handle this structure, maybe there's none. But if nothing else, the cyclic dependency could be "demoted" to a warning and the recursed entity analyzed as a black box. This would allow analysis of the IPs that are instantiating the recursive entity.

Schottkyc137, 2024-02-22:
Could you provide a minimal reproducible example? The code in the answer doesn't cause an error for me

Schottkyc137, 2025-05-04:
Likely closed with [35072bc](https://github.com/VHDL-LS/rust_hdl/commit/35072bc4059c2ad34d519ef0e18780f8cc702579). If not, please re-open this issue

closed 2025-05-04 by Schottkyc137