VHDL-LS/rust_hdl #273: Generic Library Inclusion

SethGower, 2024-03-04:
It would be nice to be able to arbitrarily include library directories if they're in some defined structure. Not sure how this would be done in the `vhdl_ls.toml`, but say you had something like this in the project
```
└── libs
    ├── library_a
    │   ├── sim
    │   └── src
    │       └── library_a.vhd
    └── library_b
        ├── sim
        └── src
            └── library_b.vhd

7 directories, 2 files
```

It would be great if it could arbitrarily pull the names `library_a` and `library_b` as libraries. This could maybe be done with a new section in the `toml` config file, where you define a directory in which it should look recursively. Maybe Establish a regex pattern with a capture group that can pull out the library name, for example with the expression `(libs)\/(?<lib>\w+).+` testing [here](https://rustexp.lpil.uk/) with this input
```
libs/library_a/src

libs/library_b/src

other/test/src
```

This would be handy if the project has (or you have globally) a repository of libraries where you have modules. This way you wouldn't need to specify all of them when you create the `vhdl_ls.toml` for the project, and you wouldn't need to update when you add new ones.

Schottkyc137, 2024-03-24:
My thoughts on this:

I'm generally in favour of having a more modular and flexible system with the `toml` files. However, I also would like to keep these files as simple as possible. In most occasions, this is likely the second configuration system people will have to maintain when working with VHDL code. There probably is another build step for example plain vendor tools (like Vivado, Quartus, ...), custom build files, makefiles, or others.
Therefore, I generally do not want to put too much logic into these files.

Something that I could get behind, however, is the idea of includes (which was already discussed in #151).
Such modifications shouldn't be too difficult to implement as some functionality already exists. For example, one can already programatically merge two config files.

SethGower, 2024-03-24:
Yeah, this makes sense. Honestly, I had made this request for a case I came up with at work, but ended up automating the updating of the `vhdl_ls.toml` file with some automation we already had in place.

I think that includes would definitely be a better way to handle this, since I feel like the original functionality I suggested in this issue would be a lot more complicated at runtime and might cause some lag-time during the LSP startup time. Personally, I use `neovim`, so I startup the LSP repeatedly during my work.

I am going to close this issue, so it isn't duplicated with #151, since  I think that is a better idea.

closed 2024-03-24 by SethGower