VHDL-LS/rust_hdl #274: Getting errors for `ieee2008/float_generic_pkg-body.vhdl`

pintert3, 2024-03-06:
Seems like the recent update broke something in the server itself. I'm getting a bunch of *No declaration of <name>* for `float_generic_pkg-body.vhdl` in my VSCode output. It looks like it's related to the server. I think most of the missing/undeclared names are from the `fixed_generic_pkg.vhdl` file that sits in the same folder.

[image]

Schottkyc137, 2024-03-06:
Note: This was fixed with  #272 but no released so far. I'll publish a new release with the fix today

pintert3, 2024-03-06:
Oh yes, I see. I think I'll close this too then.

Schottkyc137, 2024-03-06:
Should be fixed with 0.78.1