VHDL-LS/rust_hdl #278: [BUG] File paths with $ are incorrectly being interpreted as environment variables

Matt-D-Smith, 2024-03-18:
If a file path has a '$' character in it, the parser for vhdl_ls.toml interprets it as an environment variable when it should not. $ characters are valid in windows file path names, and used in network/shared file paths to indicate hidden folders.

Example entry in vhdl_ls.toml
```
unisim.files = [ 
 
 
 
 '\\networklocation\cad$\apps\xilinx_vitis\Vivado_2020.2\Vivado\2020.2\data\vhdl\src\unisims\unisim_VCOMP.vhd',
]```

Resulting error in VHDL LS output:
```
Error loading vhdl_ls.toml: environment variable '' is not defined
```

Schottkyc137, 2024-03-23:
This is fixed with the latest commit by disabling the substitution feature for windows machines. However the current solution is unsatisfactory as windows users would likely also want to have this feature, probably using the `%VAR_NAME%` syntax instead of the `$VAR_NAME` syntax. I have opened [an issue](https://github.com/fizyr/subst/issues/17) on the `subst` package addressing this.

closed 2024-03-23 by Schottkyc137

Xcodo, 2024-03-23:
Yes I'm quite disappointed by that change - the variable substitution
feature was fantastic for us and we're on Windows.

Perhaps an alternative would be to have a setting to enable or disable the
substitution?

PR #280 (2024-03-23): Temporary Fix: Do not substitute environment variables on windows
Also replace the custom implementation with [subst](https://docs.rs/subst/latest/subst/)

Fixes #278