VHDL-LS/rust_hdl #289: Autocompletion in concurrent statements broken (VSCode)

AndrzejKowalski9917, 2024-04-16:
**Description**
When trying to write a concurrent statement outside of a process, the autocompletion only suggests instantiation templates of other entities instead of signals declared within the architecture.

**How to reproduce**
I have a simple entity / architecture in one file (simple_module.vhdl) like this:
```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

entity simple_module is
  port (
    clk   : in std_logic;
    reset : in std_logic

  );
end entity;

architecture rtl of simple_module is

begin

end architecture;
```

I have a top module like this:
```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

entity top_module is
  port (
    clk   : in std_logic;
    reset : in std_logic
  );
end entity;

architecture rtl of top_module is
  signal s_foo : std_logic := '0';
  signal s_bar : std_logic_vector(3 downto 0) := (others => '0');
begin

  s_foo <= s_bar(0);

end architecture;

```

The top module has 2 signals. When I want to type the line with `s_foo <= s_bar(0);`, I start by typing `s_` and expect autocomplete to suggest the signals `s_foo` and `s_bar`. Instead only a instantiation template for the `module simple_module` is given.

[image]

**Versions used**
* I tried with vhdl_ls 0.78.1 (which is the default loaded by the VSCode extension at the moment) and the last release 0.79.0 by loading it manually (for some reason The VSCode extension did not load the last version until now).
* VSCode
* Windows 11

Schottkyc137, 2024-04-16:
At the moment, autocompletion is only implemented for entity instantiation, `use` statements and map aspects. This is the expected behaviour as of now. I'll still leave this issue open as request for signal instantiation

AndrzejKowalski9917, 2024-04-17:
Would it in the meantime be possible to additional show the signals in the suggestion popup as simple text completions as it was before? Here is a screenshot with VHDL-LS 0.68.0 which just showed the intellisense text completion (as it also does now within processes). This saves a lot of typing for concurrent statements.

[image]

Schottkyc137, 2024-04-17:
I'm not quite sure whether this is really something we can do... We have not disabled the default text-based intellisense from our side (at least as far as I know). Maybe a solution is to add an option to the language server to enable or disable the built-in completion? It might also be simple to just add the signals from the immediate scope to the things that the language server completes.

AndrzejKowalski9917, 2024-04-17:
I think adding an option is not necessary, if the real solution is easy to implement.
In this case we can do as you suggested and leave this issue open as a feature request for signal instantiation.

PR #293 (2024-04-24): Add signal and port declarations to the list of completion items
Closes #289
closed 2024-04-24 by Schottkyc137