VHDL-LS/rust_hdl #290: Signal of record type connected to input/output port of same record type reports a mismatch

singlemaltirish, 2024-04-16:
I have recently started using your extension - I must say that it is great. It unimaginably increases the comfort and efficiency of work. Thanks for developing this project.

Unfortunately, I found a certain gap in functionality. Once you declare a type in package:
[image]

Then declare an entity with input port associated with this type:
[image]

and a signal which will be connected to this port (all based on same type):
[image]

At the end once the connection to port is done LS is reporting an error, that the record type doesn't match:
[image]

Could you please take a look at it? It would be nice if this could be resolved.

Schottkyc137, 2024-04-16:
Hi, great to see that you like our project!
Unfortunately, I cannot seem to reproduce your error.
Note that it's best if you provide a minimal reproducible example (ideally as code instead of screenshots) so that we can investigate the issue faster. The example that I used is the following:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

package test is
    type record_PWMmod_3_train is record
        Pulse_Train_T_on : std_logic_vector(15 downto 0);
        Pulse_Train_Period : std_logic_vector(15 downto 0);
        Pulse_Train_Repetition : std_logic_vector(15 downto 0);
        Pulse_Train_BreakTime : std_logic_vector(15 downto 0);
    end record;

    type record_Package is array (19 downto 0) of record_PWMmod_3_train;

    type record_type_PWMmod3_package is record
        Train: record_Package;
        Pulse_Energy_To_High : std_logic;
        number_of_Train : std_logic_vector(15 downto 0);
        start_Package : std_logic;
        Pulse_Train_Active : std_logic_vector(19 downto 0);
    end record;
end package;

use work.test.all;

entity test_ent is
    port ( train : in record_type_PWMmod3_package );
end entity;

use work.test.all;

entity test_ent2 is
end entity test_ent2;

architecture rtl of test_ent2 is
    signal rec_PWMmod_3_Package : record_type_PWMmod3_package;
begin

    test_ent_inst: entity work.test_ent
     port map(
        train => rec_PWMmod_3_Package
    );
end architecture;
```

Is this similar to the setup that you are using? Does this example also cause the issue for you?

singlemaltirish, 2024-04-17:
Hi. Thank you for your rapid response. I've digged deeper into this problem. Turned out it is not really an issue from your side. I was handed an old university project in which someone clearly did not pay attention to proper library management. I managed to reproduce the issue in this minimal example:
[github_issue.zip](https://github.com/VHDL-LS/rust_hdl/files/15007092/github_issue.zip)

As you can see I've duplicated declaration of test_record_pkg and test_module both in work library and global library (see vhdl_ls.toml). 

Once I will try to instantiate  test_module from work library it is throwing the mismatch error (even though the navigation of the declaration points to the same record type)

Once I will do it properly, and throw away work library declaration from .toml and refer to test_module as a part of global library it is working completly fine. 

Strangely it is not reporting the "record already declared" issue which is present when you duplicate the declaration.

singlemaltirish, 2024-04-17:
By the way, does vhdl_ls support time unit definitions in any way? I have a constant declared as:

C_MAX_PERIOD_TIME_W              :integer := SizeSLV(1sec/20ns); -- 50_000_000

[image]

closed 2024-09-02 by Schottkyc137