VHDL-LS/rust_hdl #291: Unused lint not available in vhdl_lang

Xcodo, 2024-04-17:
I really like the new control over lint severities and am looking at adding linting to our CI using the `vhdl_lang` executable.

However, the unused declaration detection has a special case applied to it which seems to override the `[lint]` settings.

Example `vhdl_ls.toml`:
```toml
...
[lint]
unused = 'error'
```

Example file (not set as a library file)
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity UnusedExample is
    port (
        In1_i: in std_logic;
        In2_i: in std_logic;
        Out_o: out std_logic
    );
end UnusedExample;

architecture behavioural of UnusedExample is
    signal used_s: std_logic;
    signal unused_s: std_logic;
begin
    used_s <= In1_i or In2_i;
    Out_o <= used_s;
end behavioural;
```

In VSCode the error is flagged correctly (and the severity has been upgraded to error)
[image]
But running `vhdl_lang` gives no diagnostics.

I'm using v0.79.0.

I suggest that the special case for disabling unused declaration detection when using `vhdl_lang` is removed as it can not be ignored within the configuration file now.

Schottkyc137, 2024-04-18:
The project, when called from the command line, currently does not perform any unused declaration detection.
@kraigher is there a particular reason for this or was this just overlooked? I don't see a compelling reason at the moment not to enable this.

kraigher, 2024-04-19:
The vhdl_lang executable was never really created by me for external users. I added it to be able to quickly check that VHDL LS would cleanly work on real world projects in the example projects. When doing this I did not want to be spammed with unused declarations commonly found in real world projects, I only caree about avoiding false positive errors. But checking for unused declarations could have been an opt out option for vhdl_lang instead of the default.

I think vhdl_lang needs a declared purpose that it can be designed to fit. Maybe its purpose should be running checks for CI as its main role in the future.

Schottkyc137, 2024-04-20:
I guess it makes sense. Maybe there should then also be the option to output machine-parsable code (e.g., json) for better integration with pipelines. Unsure if there is a standardised format.
We could also use this in one of our pipelines as integration test to check that the example project parses without errors. This would have been helpful in a couple of PR's where the unit tests didn't catch the regression

PR #306 (2024-05-25): Refactor vhdl lang main
Closes #291

- The `vhdl_lang` executable now checks for unused declarations.
- Some arguments were removed from the executable as there are better options (i.e., benchmarking for the `perf` or `bench` option)
- Add integration test

Additionally, `vhdl_ls` will exit with exit code '1', when there are any errors found in the executable.
closed 2024-05-25 by Schottkyc137