VHDL-LS/rust_hdl #292: Configurable lint settings for third party libraries

Xcodo, 2024-04-17:
When using `vhdl_lang` for CI, it is not possible to set severities separately for different libraries.

I have several third party libraries where there are some diagnostics detected, mostly `unnecessary_work_library`. I would like to disable these diagnostics for these third party libraries, but keep them active for our own files.

Could we extend the `lint` section to have two levels of configuration, one for first party and one for third party libraries?

Xcodo, 2024-04-17:
This is also related to #291, as this setting is likely to be different for first and third party libraries.

Xcodo, 2024-04-17:
Thinking about this further, another potential solution would be to have another severity level which could be set error-by-error which would be a warning for first party libraries, but ignored for third party libraries. Something like
- Error
- Warning
- IgnoreLibraries
- False

Schottkyc137, 2024-05-18:
Probably the first step is to move stuff that should be lint (instead of regular diagnostics) in the lint section.
The two diagnostics that fall into this category are the `unnecessary_work_library` and the `unassociated_context` diagnostics. Specifying `is_third_party` would then disable these diagnostics.
Later, I plan to support per-library lint tables. Then, the `is_third_party` flag would basically be an alias to disable the lint diagnostics for these libraries.
I'm not a huge fan of adding another severity as that makes the scope of these severities rather fuzzy.