VHDL-LS/rust_hdl #295: Neovim on Mac OS cannot find vhdl_library exception and VHDL LS doesn't function

FredKellerman, 2024-04-20:
Hi,  I am using Neovim 0.9.5 Release on a Macbook running Sonoma 14.4.1 and Brew.

I used Mason to install rust_hdl and have the following in a lua script that is being executed without any errors itself:
```
lspconfig = require('lspconfig')
lspconfig['vhdl_ls'].setup({
  on_attach = on_attach,
  capabilities = capabilities
})
```
Everything seems to be setup correctly, :checkhealth doesn't report anything obvious but VHDL LS doesn't work when editing and :messages reports:
```
Client 1 quit with exit code 101 and signal 0
```
In the .local/state/nvim/lsp.log it has:
```
[ERROR][2024-04-20 17:08:25] .../vim/lsp/rpc.lua:734	"rpc"	".local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"thread 'main' panicked at vhdl_lang/src/config.rs:280:9:\nCouldn't find installed libraries at ../vhdl_libraries, ../../vhdl_libraries, /usr/lib/rust_hdl/vhdl_libraries, ../share/vhdl_libraries.\nnote: run with `RUST_BACKTRACE=1` environment variable to display a backtrace\n"
```
I have verified that the libraries are installed here:
```
 .local/share/nvim/mason/packages/rust_hdl/vhdl_ls-aarch64-apple-darwin/vhdl_libraries
```
I did try to copy and create sym links to where the error said it was looking but it didn't make any difference.  I can provide more details if needed.  Thank you.

srbarton43, 2024-05-01:
Getting same error on MacOS 14.1.1

Schottkyc137, 2024-05-11:
The issue seems to be that `Mason` installs the binary to `~/.local/share/nvim/mason/bin`, but the libraries are installed at `~/.local/share/nvim/mason/packages/rust_hdl/vhdl_ls-aarch64-apple-darwin/vhdl_libraries`. I think that this is a problem with the Mason installation process. Looking over [the package description](https://github.com/mason-org/mason-registry/blob/main/packages/rust_hdl/package.yaml), maybe the `share` property needs to be set to `{{source.asset.vhdl_libraries}}` or similar?
I can verify that symlinking the libraries to `~/.local/share/nvim/mason/share` works and causes the language server to start.
Please note that VHDL-LS does not maintain the `rust_hdl` Mason package. Therefore, as this seems to be a bug / missing installation instructions on Mason, I will not consider this a bug here.

Shivlocin, 2024-06-16:
https://github.com/mason-org/mason-registry/pull/6140

closed 2024-08-08 by Schottkyc137