VHDL-LS/rust_hdl #298: [BUG] Warning reported regarding unused generic procedures/functions

nselvara, 2024-04-27:
Hi guys, I get this warning when I use generic functions/procedures: 
`Unused declaration of procedure check_equal_generic[type_t, type_t, STRING, log_level_t, NATURAL, NATURAL, STRING]vhdl ls(unused)`

I get the reason behind why it's flagged as not used but I think it should only emit this warning when it hasn't being defined with a generic map.

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library vunit_lib;
context vunit_lib.vunit_context;

library osvvm;
use osvvm.NamePkg.NamePType;

package pkg_test is
    type test_t is (test_1, test_2, test_3);

    procedure check_equal_generic
        generic (
            type type_t;
            function to_string(data: type_t) return NamePType
        )
        parameter (
            got: in type_t;
            expected: in type_t;
            msg: in string := check_result_tag;
            level: in log_level_t := null_log_level;
            path_offset: in natural := 1;
            line_num: in natural := 0;
            file_name: in string := ""
        );

    procedure check_equal is new check_equal_generic generic map (data_t => test_t);
end package;

package body pkg_test is
    function to_string(data: test_t) return NamePType is
        variable result: NamePType;    
    begin
        case data is
            when test_1 => result.Set("test_1");
            when test_2 => result.Set("test_2");
            when test_3 => result.Set("test_3");
        end case;
        return result;
    end function;

    procedure check_equal_generic
        generic (
            type type_t;
            function to_string(data: type_t) return NamePType
        )
        parameter (
            got: in type_t;
            expected: in type_t;
            msg: in string := check_result_tag;
            level: in log_level_t := null_log_level;
            path_offset: in natural := 1;
            line_num: in natural := 0;
            file_name: in string := ""
        ) is
        constant checker: checker_t := default_checker;
    begin
        -- pragma translate_off
        if got = expected then
            if is_pass_visible(checker) then
                passing_check(
                    checker,
                    p_std_msg(
                        "Equality check passed", msg,
                        "Got " & to_string(got).Get & "."
                    ),
                    path_offset + 1, line_num, file_name
                );
            else
                passing_check(checker);
            end if;
        else
            failing_check(
                checker,
                p_std_msg(
                    "Equality check failed", msg,
                    "Got " & to_string(got).Get & ". " &
                    "Expected " & to_string(expected).Get & "."
                ),
                level, path_offset + 1, line_num, file_name
            );
        end if;
    end;
end package body;
```

Schottkyc137, 2024-04-28:
Can confirm. Here's a shorter MRE
```vhdl
package pkg_test is
    procedure foo
        generic (type type_t)
        parameter (bar: type_t);

    procedure baz is new foo generic map (type_t => natural);
end package;

package body pkg_test is
    procedure foo
        generic (type type_t)
        parameter (bar: type_t)
    is begin
    end;
end package body;
```

The actual problem is that the information that the `foo` procedure implementation is linked to the `foo` procedure specification is not generated correctly at the moment (reference : https://github.com/VHDL-LS/rust_hdl/blob/master/vhdl_lang/src/analysis/subprogram.rs#L424).

nselvara, 2024-04-29:
Hihi, yes the shorter one is more preferable to analyse. 😸