VHDL-LS/rust_hdl #299: [Clarification] No warning reported if object is not set

nselvara, 2024-04-29:
As an addendum to this issue: https://github.com/VHDL-LS/rust_hdl/issues/298
If one doesn't use/set the object (port, signal, variable), VHDL-LS correctly flags a warning that it's not set - all good :) .
However, if the object is used for example as `bla'sig_obj'subtype` then the warning goes away even though the object is kind of not set. I'm not sure if that's a desirable feature to also output warning if the signal is not set rather than just used to get meta data.

Schottkyc137, 2024-05-03:
It's a good question whether this should be marked as an unused declaration or not. `rust-analyzer` does something similar when a variable is only used in recursion, so maybe `vhdl_ls` should emit an error in a similar fashion?

nselvara, 2024-05-05:
Or maybe, we introduce 2 kind of flags, which then indicates that the object is only used as a reference and the other one remains the same if it's not used.

nselvara, 2024-06-25:
I found another one that's related to this:

This one flags that the index is not used.
```vhdl
variable bla: boolean;

function ret_bla(index: natural) return boolean is begin
  return true;
end function;

bla := ret_bla(0);
```

However, if you write the usage like this:

```vhdl
bla := ret_bla(index => 0);
```

The warning goes away even though the index is not used really.