VHDL-LS/rust_hdl #300: Panic when (seemingly) completing a port name

SethGower, 2024-05-02:
Hello. It seems that the LSP is panicking and exiting with exit code `101` when I try to complete a port name.

Minimal example:
`vhdl_ls` version `v0.80.0`
Client: Neovim with latest [`nvim-lspconfig`](https://github.com/neovim/nvim-lspconfig), LSP installed with [`mason`](https://github.com/williamboman/mason.nvim).  Using the default configuration in `nvim-lspconfig`, but [here](https://github.com/SethGower/dotfiles/blob/main/config/nvim/lua/plugins/lsp.lua#L79) is my config for the LSP.

`vhdl_ls.toml`
```toml
[libraries]
defaultlib.files = [
    "./*.vhd"
]
```

VHDL File
```vhdl
library ieee;
    use ieee.std_logic_1164.all;
    use ieee.numeric_std.all;

entity test1 is
    port (
        i_clk  : in    std_logic;
        i_rstn : in    std_logic;
        i_d    : in    std_logic;
        o_q    : out   std_logic;
        o_qn   : out   std_logic
    );
end entity test1;

architecture behav of test1 is

begin

    o_q_proc : process (i_clk) is
    begin

        if rising_edge(i_clk) then
            if (i_rstn = '0') then
                o_q <= '0';
            else
                o_q <= i_d;
            end if;
        end if;

    end process o_q_proc;

end architecture behav;
```

If I try to add `o_qn <= not o_q`, by typing out `o_` and then selecting the autocomplete candidate from the LSP, I get the crash. Seems to be a rust panic. I have attached the log for the whole LSP session. The only related line seems to be the last one, however I figured I'd include it all
```
[START][2024-05-02 15:48:41] LSP logging initiated
[INFO][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:662	"Starting RPC client"	{  args = {},  cmd = "/home/sgower/.local/share/nvim/mason/bin/vhdl_ls",  extra = {    cwd = "/home/sgower/Documents/Development/test/vhdl"  }}
[DEBUG][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:284	"rpc.send"	{  id = 1,  jsonrpc = "2.0",  method = "initialize",  params = {    capabilities = {      textDocument = {        callHierarchy = {          dynamicRegistration = false        },        codeAction = {          codeActionLiteralSupport = {            codeActionKind = {              valueSet = { "", "quickfix", "refactor", "refactor.extract", "refactor.inline", "refactor.rewrite", "source", "source.organizeImports" }            }          },          dataSupport = true,          dynamicRegistration = false,          isPreferredSupport = true,          resolveSupport = {            properties = { "edit" }          }        },        completion = {          completionItem = {            commitCharactersSupport = false,            deprecatedSupport = false,            documentationFormat = { "markdown", "plaintext" },            preselectSupport = false,            resolveSupport = {              properties = { "documentation", "detail", "additionalTextEdits" }            },            snippetSupport = true          },          completionItemKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25 }          },          contextSupport = false,          dynamicRegistration = false        },        declaration = {          linkSupport = true        },        definition = {          linkSupport = true        },        documentHighlight = {          dynamicRegistration = false        },        documentSymbol = {          dynamicRegistration = false,          hierarchicalDocumentSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        hover = {          contentFormat = { "markdown", "plaintext" },          dynamicRegistration = false        },        implementation = {          linkSupport = true        },        publishDiagnostics = {          relatedInformation = true,          tagSupport = {            valueSet = { 1, 2 }          }        },        references = {          dynamicRegistration = false        },        rename = {          dynamicRegistration = false,          prepareSupport = true        },        semanticTokens = {          augmentsSyntaxTokens = true,          dynamicRegistration = false,          formats = { "relative" },          multilineTokenSupport = false,          overlappingTokenSupport = true,          requests = {            full = {              delta = true            },            range = false          },          serverCancelSupport = false,          tokenModifiers = { "declaration", "definition", "readonly", "static", "deprecated", "abstract", "async", "modification", "documentation", "defaultLibrary" },          tokenTypes = { "namespace", "type", "class", "enum", "interface", "struct", "typeParameter", "parameter", "variable", "property", "enumMember", "event", "function", "method", "macro", "keyword", "modifier", "comment", "string", "number", "regexp", "operator", "decorator" }        },        signatureHelp = {          dynamicRegistration = false,          signatureInformation = {            activeParameterSupport = true,            documentationFormat = { "markdown", "plaintext" },            parameterInformation = {              labelOffsetSupport = true            }          }        },        synchronization = {          didSave = true,          dynamicRegistration = false,          willSave = true,          willSaveWaitUntil = true        },        typeDefinition = {          linkSupport = true        }      },      window = {        showDocument = {          support = true        },        showMessage = {          messageActionItem = {            additionalPropertiesSupport = false          }        },        workDoneProgress = true      },      workspace = {        applyEdit = true,        configuration = true,        didChangeWatchedFiles = {          dynamicRegistration = false,          relativePatternSupport = true        },        semanticTokens = {          refreshSupport = true        },        symbol = {          dynamicRegistration = false,          hierarchicalWorkspaceSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        workspaceEdit = {          resourceOperations = { "rename", "create", "delete" }        },        workspaceFolders = true      }    },    clientInfo = {      name = "Neovim",      version = "0.9.5"    },    initializationOptions = vim.empty_dict(),    processId = 2692421,    rootPath = "/home/sgower/Documents/Development/test/vhdl",    rootUri = "file:///home/sgower/Documents/Development/test/vhdl",    trace = "off",    workspaceFolders = { {        name = "/home/sgower/Documents/Development/test/vhdl",        uri = "file:///home/sgower/Documents/Development/test/vhdl"      } }  }}
[DEBUG][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:387	"rpc.receive"	{  jsonrpc = "2.0",  method = "window/logMessage",  params = {    message = "Loaded Installation configuration file: /home/sgower/.local/share/nvim/mason/packages/rust_hdl/vhdl_ls-x86_64-unknown-linux-musl/bin/../vhdl_libraries/vhdl_ls.toml",    type = 4  }}
[DEBUG][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:387	"rpc.receive"	{  jsonrpc = "2.0",  method = "window/logMessage",  params = {    message = "Loaded HOME folder configuration file: /home/sgower/.vhdl_ls.toml",    type = 4  }}
[DEBUG][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:387	"rpc.receive"	{  jsonrpc = "2.0",  method = "window/logMessage",  params = {    message = "Loaded workspace root configuration file: /home/sgower/Documents/Development/test/vhdl/vhdl_ls.toml",    type = 4  }}
[INFO][2024-05-02 15:48:41] ...lsp/handlers.lua:539	"Loaded Installation configuration file: /home/sgower/.local/share/nvim/mason/packages/rust_hdl/vhdl_ls-x86_64-unknown-linux-musl/bin/../vhdl_libraries/vhdl_ls.toml"
[INFO][2024-05-02 15:48:41] ...lsp/handlers.lua:539	"Loaded HOME folder configuration file: /home/sgower/.vhdl_ls.toml"
[INFO][2024-05-02 15:48:41] ...lsp/handlers.lua:539	"Loaded workspace root configuration file: /home/sgower/Documents/Development/test/vhdl/vhdl_ls.toml"
[DEBUG][2024-05-02 15:48:41] .../vim/lsp/rpc.lua:387	"rpc.receive"	{  id = 1,  jsonrpc = "2.0",  result = {    capabilities = {      completionProvider = {        completionItem = {          labelDetailsSupport = true        },        resolveProvider = true,        triggerCharacters = { "." }      },      declarationProvider = true,      definitionProvider = true,      d [... cut ...]
[ERROR][2024-05-02 15:48:57] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"thread 'main' panicked at vhdl_lang/src/named_entity/arena.rs:113:32:\nno entry found for key\nnote: run with `RUST_BACKTRACE=1` environment variable to display a backtrace\n"
```

Seems to be pointing to an error related the the dict not having a key here https://github.com/VHDL-LS/rust_hdl/blob/56f17db69a56b0e094667a9da83fb17cd20922c2/vhdl_lang/src/named_entity/arena.rs#L113

Schottkyc137, 2024-05-03:
Is the bug reproducible? I.e., does what you describe always happen?
I cannot reproduce the bug on VSCode, could that be something that only affects Neovim? (Not trying to shift the blame to somewhere else, just trying to reproduce it)

SethGower, 2024-05-08:
I was able to reproduce this consistently, given the example I put in the issue. Maybe it could be a neovim thing? I can try in VSCode, however I don't use that much, so I'll need to get it setup

SethGower, 2024-05-08:
I was unable to get it to reproduce in VSCode, however as I said, I can consistently get it to reproduce in neovim.

It would be hard to isolate where the issue resides (in the client vs server) since the feature that seems to be causing the issue was adding in v0.80. However, I just tried backing up my plugin environment to before I updated a lot of plugins in neovim on Thursday (which I think I did before experiencing this bug) to see if that solves it. When I had all of the old versions of the plugins loaded, I still get the issue.

SethGower, 2024-05-08:
Digging more, and I noticed that it doesn't seem to be happening when I use the [`minimal_init.lua`](https://github.com/neovim/nvim-lspconfig/blob/master/test/minimal_init.lua) that `nvim-lspconfig` provides, when using the "Omnifunc" completion. So this could be related to my autocompletion setup? I'll dig more

SethGower, 2024-05-09:
Quick update: The panic doesn't happen when selecting a completion candidate when filling a port map of an instantiation. 

So if in another module I were instantiating my example entity described above, if I typed `o_|` (where `|` is my cursor), it would give me options including `o_q` and `o_qn`, if I hit Tab to select one, it doesn't crash. However if I try to select a signal/port within the same module, it does crash.

SethGower, 2024-05-10:
Did some digging with the above observation. Here's the log for when I first try and complete a port in the port map when instantiating an object (which succeeds), then I try to associate it with a completion to a local signal/port (which fails)
<details>

<summary> :LspLog output for both completions (good and then bad) </summary>

```
[START][2024-05-10 11:50:11] LSP logging initiated
[INFO][2024-05-10 11:50:11] .../vim/lsp/rpc.lua:662	"Starting RPC client"	{  args = {},  cmd = "/home/sgower/.local/share/nvim/mason/bin/vhdl_ls",  extra = {    cwd = "/home/sgower/Documents/Development/test/vhdl"  }}
[DEBUG][2024-05-10 11:50:11] .../vim/lsp/rpc.lua:284	"rpc.send"	{  id = 1,  jsonrpc = "2.0",  method = "initialize",  params = {    capabilities = {      textDocument = {        callHierarchy = {          dynamicRegistration = false        },        codeAction = {          codeActionLiteralSupport = {            codeActionKind = {              valueSet = { "", "quickfix", "refactor", "refactor.extract", "refactor.inline", "refactor.rewrite", "source", "source.organizeImports" }            }          },          dataSupport = true,          dynamicRegistration = false,          isPreferredSupport = true,          resolveSupport = {            properties = { "edit" }          }        },        completion = {          completionItem = {            commitCharactersSupport = false,            deprecatedSupport = false,            documentationFormat = { "markdown", "plaintext" },            preselectSupport = false,            resolveSupport = {              properties = { "documentation", "detail", "additionalTextEdits" }            },            snippetSupport = true          },          completionItemKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25 }          },          contextSupport = false,          dynamicRegistration = false        },        declaration = {          linkSupport = true        },        definition = {          linkSupport = true        },        documentHighlight = {          dynamicRegistration = false        },        documentSymbol = {          dynamicRegistration = false,          hierarchicalDocumentSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        hover = {          contentFormat = { "markdown", "plaintext" },          dynamicRegistration = false        },        implementation = {          linkSupport = true        },        publishDiagnostics = {          relatedInformation = true,          tagSupport = {            valueSet = { 1, 2 }          }        },        references = {          dynamicRegistration = false        },        rename = {          dynamicRegistration = false,          prepareSupport = true        },        semanticTokens = {          augmentsSyntaxTokens = true,          dynamicRegistration = false,          formats = { "relative" },          multilineTokenSupport = false,          overlappingTokenSupport = true,          requests = {            full = {              delta = true            },            range = false          },          serverCancelSupport = false,          tokenModifiers = { "declaration", "definition", "readonly", "static", "deprecated", "abstract", "async", "modification", "documentation", "defaultLibrary" },          tokenTypes = { "namespace", "type", "class", "enum", "interface", "struct", "typeParameter", "parameter", "variable", "property", "enumMember", "event", "function", "method", "macro", "keyword", "modifier", "comment", "string", "number", "regexp", "operator", "decorator" }        },        signatureHelp = {          dynamicRegistration = false,          signatureInformation = {            activeParameterSupport = true,            documentationFormat = { "markdown", "plaintext" },            parameterInformation = {              labelOffsetSupport = true            }          }        },        synchronization = {          didSave = true,          dynamicRegistration = false,          willSave = true,          willSaveWaitUntil = true        },        typeDefinition = {          linkSupport = true        }      },      window = {        showDocument = {          support = true        },        showMessage = {          messageActionItem = {            additionalPropertiesSupport = false          }        },        workDoneProgress = true      },      workspace = {        applyEdit = true,        configuration = true,        didChangeWatchedFiles = {          dynamicRegistration = false,          relativePatternSupport = true        },        semanticTokens = {          refreshSupport = true        },        symbol = {          dynamicRegistration = false,          hierarchicalWorkspaceSymbolSupport = true,          symbolKind = {            valueSet = { 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26 }          }        },        workspaceEdit = {          resourceOperations = { "rename", "create", "delete" }        },        workspaceFolders = true      }    },    clientInfo = {      name = "Neovim",      version = "0.9.5"    },    initializationOptions = vim.empty_dict(),    processId = 2594612,    rootPath = "/home/sgower/Documents/Development/test/vhdl",    rootUri = "file:///home/sgower/Documents/Development/test/vhdl",    trace = "off",    workspaceFolders [... cut ...]
        o_leds     : out   std_logic_vector(7 downto 0)
    );
end entity test2;

architecture behav of test2 is

    signal s_leds_not : std_logic_vector(7 downto 0);

begin

    dff_generate : for i in 0 to 7 generate

        test1_0 : entity work.test1
            port map (
                i_clk  => i_clk,
                i_rstn => '1',
                i_d    => i_switches(i),
                o_q    => o_leds(i),
                o_qn   => not o_leds
            );

    end generate dff_generate;

end architecture behav;
```

SethGower, 2024-05-10:
Sorry I am sort of blowing this issue up, hope I'm not spamming your inboxes too much. If discussion should be moved elsewhere, let me know

Here are more detailed error outputs

<details>

<summary>RUST_BACKTRACE=1</summary>

```
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"thread 'main' panicked at vhdl_lang/src/named_entity/arena.rs:113:32:\nno entry found for key\nstack backtrace:\n"
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   0: rust_begin_unwind\n             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/panicking.rs:647:5\n   1: core::panicking::panic_fmt\n             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/panicking.rs:72:14\n   2: core::panicking::panic_display\n             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/panicking.rs:196:5\n   3: core::panicking::panic_str\n             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/panicking.rs:171:5\n   4: core::option::expect_failed\n             at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/option.rs:1988:5\n"
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   5: vhdl_lang::named_entity::arena::FinalArena::get\n"
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   6: vhdl_lang::project::Project::format_entity\n"
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   7: vhdl_ls::vhdl_server::VHDLServer::resolve_completion_item\n   8: vhdl_ls::stdio_server::ConnectionRpcChannel::main_event_loop\n   9: vhdl_ls::stdio_server::start\n  10: vhdl_ls::main\n"
[ERROR][2024-05-10 12:07:58] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.\n"
```

</details>

<details>

<summary>RUST_BACKTRACE=full</summary>

```
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"thread 'main' panicked at vhdl_lang/src/named_entity/arena.rs:113:32:\nno entry found for key\nstack backtrace:\n"
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   0:     0x78e95646ffa6 - std::backtrace_rs::backtrace::libunwind::trace::h892d1706cd1db3e2\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5\n   1:     0x78e95646ffa6 - std::backtrace_rs::backtrace::trace_unsynchronized::h3ecd625736285bad\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5\n   2:     0x78e95646ffa6 - std::sys_common::backtrace::_print_fmt::h4d59e1a015be3f6a\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sys_common/backtrace.rs:"
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"68:5\n   3:     0x78e95646ffa6 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h22749e6c189a1ba4\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sys_common/backtrace.rs:44:22\n"
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"   4:     0x78e9564add10 - core::fmt::rt::Argument::fmt::h628b62d266085cbe\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/fmt/rt.rs:142:9\n   5:     0x78e9564add10 - core::fmt::write::hbd839685e035b696\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/core/src/fmt/mod.rs:1120:17\n   6:     0x78e95646d28f - std::io::Write::"
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"write_fmt::hf6c409a0c9571fd5\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/io/mod.rs:1846:15\n   7:     0x78e95646fd84 - std::sys_common::backtrace::_print::hb51728f3c026694b\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sys_common/backtrace.rs:47:5\n   8:     0x78e95646fd84 - std::sys_common::backtrace::print::hb846384473985f19\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/sys_common/backtrace.rs:34:9\n   9:     0x78e956471607 - std::panicking::default_hook::{{closure}}::h76c8448fbeb8f025\n"
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"  10:     0x78e956471369 - std::panicking::default_hook::h3e223b06c1476522\n            "
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"                   at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/panicking.rs:292:9\n"
[... cut ...] wind::hb922373e2e28a400\n                               at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/panic.rs:142:14\n  35:     0x78e956464908 - std::rt::lang_start_internal::h25c4055374a24228\n "
[ERROR][2024-05-10 12:08:30] .../vim/lsp/rpc.lua:734	"rpc"	"/home/sgower/.local/share/nvim/mason/bin/vhdl_ls"	"stderr"	"                              at /rustc/25ef9e3d85d934b27d9dada2f9dd52b1dc63bb04/library/std/src/rt.rs:148:20\n  36:     0x78e95605eb65 - main\n"
```

</details>

Schottkyc137, 2024-05-10:
> Sorry I am sort of blowing this issue up, hope I'm not spamming your inboxes too much. If discussion should be moved elsewhere, let me know
> 
> Here are more detailed error outputs
> 
> RUST_BACKTRACE=1
> RUST_BACKTRACE=full

Don't worry about spamming my inbox, your efforts are greatly appreciated. I suspect that somehow the Entity Id gets lost in translation. This might be an issue with vhdl_ls, or the server.
When completing items, servers can attach some data that the client will echo to request more information. For some reason, in this echo process, the entity id sent back from the client is no longer valid. The relevant code can be found [here](https://github.com/VHDL-LS/rust_hdl/blob/2a82770de3f54b2469be3f524de3d2ee35cd2ccc/vhdl_ls/src/vhdl_server.rs#L463). Maybe it's worth adding a debug statement to see the entity ID echoed by the client and the entities in the pool [here](https://github.com/VHDL-LS/rust_hdl/blob/2a82770de3f54b2469be3f524de3d2ee35cd2ccc/vhdl_lang/src/named_entity/arena.rs#L112)?
In either way, the server should probably handle this more gracefully and not crash because some faulty ID was sent.

PR #303 (2024-05-11): Check raw entity IDs from an LSP client
Closes #300 

While this might not be the final / ideal solution, checks that a raw entity id points to a valid entity reference before returning it.
This handles situations gracefully where raw entity IDs are used to pass information.
closed 2024-05-11 by Schottkyc137