VHDL-LS/rust_hdl #312: Autocomplete for variables, records, generate statements (VSCode)

AndrzejKowalski9917, 2024-06-18:
The issue might be slightly related to #289.

Recent changes to autocomplete have disabled large parts of VSCodes text based intellisense. While the filtering of the autocomplete results of vhdl-ls is a very welcome addition, it leads to a lot of typing and name lookup, if the statement is not autocompleted by vhdl-ls.
I noticed the following examples which all worked before with text based intellisense. 

* **generate statements**  
  In generate statements neither constants for `for` loops nor the `generate` keyword itself are suggested. Only if the finale `end generate` is already present, the text based auto complete suggests constants or the `generate` keyword. Since you usually don't type `end generate` before the the rest, it's basically not working.  
  No constant completion:
  [image]  
  No `generate` completion:
  [image]  
  `generate` completion as text based completion with `end generate` already typed:
  [image]

* **variables**  
  Variables are not autocompleted by vhdl-ls and have to be typed manually:  
  [image]
  [image]

* **records**  
  Records are not autocompleted and as such after the `.` only signals are suggested. 
  The port suggestion works:    
  [image]
  Members of the record are not shown and have to be typed manually without text based intellisense:  
  [image]

It might be that there are more cases that I missed here.

The code used for testing is the following. If it is typed down from top to bottom without copying it, the above behavior can be observed. 

```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

package led_pkg is
  constant gc_led_cnt : positive := 8;

  type t_led_ctrl is record
    is_active : boolean;
  end record;

  constant gc_led_ctrl_init : t_led_ctrl := (
    is_active => false
  );

  type t_led_ctrl_array is array(0 to gc_led_cnt - 1) of t_led_ctrl;
  constant gc_led_array_init : t_led_ctrl_array := (others => gc_led_ctrl_init);
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;
use work.led_pkg.all;

entity leds is
  port (
    clk   : in  std_logic;
    reset : in  std_logic;
    ctrl  : in  t_led_ctrl_array;
    btns  : in  std_logic_vector(gc_led_cnt - 1 downto 0);
    leds  : out std_logic_vector(gc_led_cnt - 1 downto 0)
  );
end entity;

architecture rtl of leds is
begin

  gen_leds_proc : for i in 0 to gc_led_cnt generate
    do_stuff : process(clk) is
      variable v_edge_detect : std_logic := '0';
    begin
      if rising_edge(clk) then
        if reset = '1' then
          leds(i)       <= '0';
          v_edge_detect := '0';
        else
          if v_edge_detect and not btns(i) then
            if ctrl(i).is_active then
              leds(i) <= '1';
            end if;
          end if;
          v_edge_detect := btns(i);
        end if;
      end if;
    end process;
  end generate;

  gen_test : for i in 0 to gc_led_cnt generate

  end generate;
end architecture;
```

AndrzejKowalski9917, 2024-07-02:
First of all, thanks for keeping to improve this already awesome project!
I tested with vhdl-ls 0.82.0 and found a lot of the things are working now. I wanted to share my results regarding the points I made in this issue:

- [x] **generate statements**  
     The autocompletion of local and global constants is now working
- [ ] **variables**
    Autocompletion of variables is not yet working  
- [x] **records**  
       Element completion is working.
- [x] **Call to functions / procedures**  
    Autocompletion of function and procedure calls is working
- [x] **global constants in entity instantiations (as argument for generics)**  
       This is now working.

I also very much like the autocompletion of types and conversion functions like `to_integer`. Would it be possible (maybe through the vhdl-ls.toml configuration) to make this autocompletions lower case?

A minor issue I could observe was the following:
When typing a character in single quotes (like when assigning a value to a `std_logic`), the following semicolon is autocompleted within single quotes when hitting enter, instead of smicolon + new line.
[image]
[image]

flokaa123, 2025-04-24:
Hello,
what is the current state on the autocompletion of variables? I am having issues with those and wanted to clarify if this is still up to date.

Schottkyc137, 2025-05-04:
> Hello, what is the current state on the autocompletion of variables? I am having issues with those and wanted to clarify if this is still up to date.

With limited time for this project my focus currently is mostly on fixing bugs for this project. Unfortunately, I cannot make a prediction when this feature will be implemented.