VHDL-LS/rust_hdl #322: [BUG] Auto instantiation does not appear given undetermined context

lgu-appear, 2024-07-26:
Hi, it seems that the auto instantiation from mapped entities in the vhdl_ls.toml does not work "everywhere". I had a simple entity where I was instantiated other entities. The only specific thing in that entity is that there was a block, but I was not trying to use the auto instantiation in the block. Tried moving things around, couldn't figure why it was working at some places but not at others...

Schottkyc137, 2024-07-28:
I think a minimal reproducible of your example is something like this, correct?

```vhdl
entity foo is
end entity foo;

architecture arch of foo is
begin

    bar : block
    begin
        -- No instantiation completions are available here
    end block;

end architecture;
```

Schottkyc137, 2024-07-28:
Should be closed with https://github.com/VHDL-LS/rust_hdl/commit/ff09201fdeeb5bfbf053061db3224370b114ed7e

lgu-appear, 2024-07-29:
Hey! No, in my case the instation completions didn't happen **out** of the block. Either before or after. It was not showing up, which was why I was confused for the context of reproducing that

Schottkyc137, 2024-07-29:
Hmm can you somehow create a minimal reproducible example? It's hard for me to understand what you mean at the moment unfortunately.

lgu-appear, 2024-08-01:
Hello,
I tried and I cannot reproduce it 😓 I am wondering if I forgot to include the lib in the file, but I am 90% sure I didn't fortget. Probably good to merge your fix and I will reopen an issue if I can see an odd behavior again.

Schottkyc137, 2024-09-02:
I'll close this for now assuming that the bug has been fixed with the latest release but feel free to re-open if the bug persists

closed 2024-09-02 by Schottkyc137