VHDL-LS/rust_hdl #324: vhdl-ls incorrectly shows type mismatch when using interfaces, views and package generics

nathanaelhuffman, 2024-07-29:
I have a minimal reproducer, the port map at line 46 is (incorrectly I think) labeled as an error with the message

"signal 'my_sig' of record type 'test_t' does not match record type 'test_t'" and I believe this is incorrect unless I'm misunderstanding something

```
library ieee;
use ieee.std_logic_1164.all;

package test_pkg is
    generic (
        width : integer
    );
    type test_t is record
        a : std_logic;
        b : std_logic_vector(width-1 downto 0);
    end record;
    
    view vone of test_t is
        a : in;
        b : out;
    end view;
    alias vtwo is vone'converse;
end package;

package w8_pkg is new work.test_pkg generic map (width => 8);

library ieee;
use ieee.std_logic_1164.all;
use work.w8_pkg.all;

entity test_sub_entity is
    port (
        my_if : view vone
    );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use work.w8_pkg.all;

entity test_top_entity is
end entity;

architecture rtl of test_top_entity is
    signal my_sig : test_t;
begin
    my_if : entity work.test_sub_entity
        port map (
            my_if => my_sig  -- <-- shows error here when there should not be an error
        );
end architecture;

```

nathanaelhuffman, 2024-07-30:
I know I'm out on the bleeding edge of things here wrt tooling support. I did notice similar errors when using arrays of interfaces in a portmap also, it is possible this is the same root-cause but I wanted to mention it for completeness:

`responders : view (axil8x32_pkg.axil_target) of axil8x32_pkg.axil_array_t(config_array'length - 1 downto 0)`

gives errors along these lines, which look similar.
```
Specified subtype must match the subtype declared for the viewvhdl ls(type_mismatch)
Specified subtype must match the subtype declared for the viewvhdl ls(type_mismatch)
type axil_array_t is array (natural range <>) of axil_t;
type axil_array_t is array (natural range <>) of axil_t;
```

I can work out a similar reproducer for this if this looks like a different bug and is required, but nvc accepts the code and runs just fine.

Schottkyc137, 2024-07-30:
Hi @nathanaelhuffman 

> I did notice similar errors when using arrays of interfaces in a portmap also, it is possible this is the same root-cause but I wanted to mention it for completeness:

Are you also experiencing these issues with generic packages as well? I peeked around a little and found that the root cause behind issue number 1 is actually related to how the generic package instances are created. If your second issue can be reproduced without package instantiations, this is likely a different cause and an MRE would be great.

nathanaelhuffman, 2024-07-30:
> Hi @nathanaelhuffman
>
>
> Are you also experiencing these issues with generic packages as well? I peeked around a little and found that the root cause behind issue number 1 is actually related to how [...]

I'll poke around and see. My current case is still using the generic packages but I'll find some time to verify whether or not this happens in normal cases.

Much appreciated for your continued work here and fast responses, I'm using vhdl-ls daily!

PR #334 (2024-08-18): Fix interface view package mismatch
Closes #324

Previously, instantiating a package unconditionally mapped all types. This is problematic for situations like the following:
```vhdl
package foo is
    generic (...);

    type foo is record
        ...
    end record foo;
    
    view bar of foo is
        ...
    end view bar;
end package foo;
```

Since `foo` is used as subtype, instantiating that package would also map `foo` and implicitly declare two `foo`s that are not compatible.

This PR fixes these issues by also including the current scope and taking values from the current scope wher [... cut]
closed 2024-08-18 by Schottkyc137