VHDL-LS/rust_hdl #327: File paths are returned without symlinks inside flatpaks

HendrikMennen, 2024-08-03:
I implemented this LS into https://github.com/one-ware/OneWare.

While running in flatpaks on a system that has a /home symlink to /var/home, when I open a LS with the workspace inside /home/..., the LS reports paths as /var/home/... instead of /home/...

Other language servers I tested don't have this behavior. 

This causes my IDE to not find the correct file for linting, since it expects a /home/... path and not /var/home/...

While there is a potential workaround for this, it would be better for the LS to report the file paths without resolving the symlinks.

Schottkyc137, 2024-08-03:
Hi @HendrikMennen 

So to understand you correctly, this is a feature request to disable symbolic link resolution in the language server, correct?

HendrikMennen, 2024-08-03:
I think it should be the default behavior, as it is the same with other language servers. 
The same issue happens with VSCode aswell

Schottkyc137, 2024-08-07:
As a quick follow-up: Are you using a `vhdl_ls.toml` file? If so, where is that file located (i.e., in your home folder or in the workspace root?

HendrikMennen, 2024-08-07:
I am usually using one at workspace root. I just tested with VSCode and without a vhdl_ls.toml, which gives the same result. I am not familiar with rust, but it seems like something similar to "realpath" gets uses to resolve all the symlinks

Schottkyc137, 2024-08-08:
I just found the reason to resolve symlinks. The actual issue was #115 where there seemed to be some weirdness in windows path handling, therefore [dunce::canonicalize](https://docs.rs/dunce/latest/dunce/fn.canonicalize.html) was used to resolve the issues. I think in order not to resolve symlinks, we could therefore also use [dunce::simplified](https://docs.rs/dunce/latest/dunce/fn.simplified.html#).

Schottkyc137, 2024-08-08:
I have created PR https://github.com/VHDL-LS/rust_hdl/pull/331. Would it be possible for you to check that and see whether that resolves your issue?

HendrikMennen, 2024-08-08:
I tested it, and works perfectly on my system!

Schottkyc137, 2024-08-08:
Thanks a lot for checking!

closed 2024-08-08 by Schottkyc137