VHDL-LS/rust_hdl #337: Copyright symbol error

qarlosalberto, 2024-08-27:
I remember that this problem was fixed some time ago. But it has appeared again when I have updated rust_hdl in TerosHDL. 

https://github.com/TerosTechnology/vscode-terosHDL/issues/646

The problem is related to `"Copyright © 2008 IEEE. All rights reserved."`. Could you remvoe all the © symbol? Or I can do it if it's ok for you.

Thanks!

Schottkyc137, 2024-08-27:
IMO, the "correct" solution is to require a VHDL file to be Latin-1 encoded (at least that's what the standard requires). But I agree that this is cumbersome as most editors will open files using utf-8 by default. That being said, I'm okay with you removing the copyright symbols, just open a PR.

Schottkyc137, 2024-08-27:
Closed with #338

closed 2024-08-27 by Schottkyc137