VHDL-LS/rust_hdl #339: [FEATURE] Custom VHDL libs location

maxime-lr, 2024-08-28:
It would be great to have an environment variable or an executable option to specify where the VHDL libraries are located.

maxime-lr, 2024-08-29:
It is a feature for the language server. And then, it could be a setting in rush_hdl_vscode.

Hinara, 2024-09-01:
You mean without an external file ?
Because vhdl-ls already accept an environment variable for the path of a file defining libs paths
https://github.com/VHDL-LS/rust_hdl/tree/master?tab=readme-ov-file#configuration

maxime-lr, 2024-09-02:
> You mean without an external file ? Because vhdl-ls already accept an environment variable for the path of a file defining libs paths https://github.com/VHDL-LS/rust_hdl/tree/master?tab=readme-ov-file#configuration

I was talking about VHDL libraries (like ieee).
I am not sure it is related to my feature request (correct me if I'm wrong 🙌)

Hinara, 2024-09-03:
Yeah I understand that you want to define the path of your libraries, but this can be done with a configuration toml file so for example if you use a library called `mylib` with all its vhd file for that library under `C:\mylib`
To use your library you would put that into your file
```vhdl
library mylib;
```
Then make it work you would need to create a toml file with a content like that
```
[libraries]
mylib.files = [
    'C:\mylib\*.vhd',
]
```

maxime-lr, 2024-09-04:
Yes, that is what I do already.
In this issue I was talking about standard libraries like (std and ieee). They are stored in some predefined directories and to my knowledge, it is not possible to define a generic path.
Maybe I'm wrong ;)

PR #344 (2024-09-21): Add option for the libraries location as part of the vhdl_lang executable
Introduces a flag so `vhdl_ls` can be called using `vhdl_ls --libararies /path/to/libraries`

Closes #339
closed 2024-09-21 by Schottkyc137