VHDL-LS/rust_hdl #340: Snippets Causing Huge Delay

SethGower, 2024-08-29:
Not sure if this is a great title, so I can change it if necessary. I have been having an issue recently, where I have _huge_ delays when my editor (Neovim) goes to complete something. I boiled it down to when the client says that snippets are supported or not (with the `capabilities.textDocument.completion.completionItem.snippetSupport` flag). I have further boiled it down to when there is a library in the scope that has a lot of possible entities that can be completed. The reason I am hitting this is because of Xilinx's `unisim` library, which houses the VHDL component declarations for all of their primitive elements. In this file, there are a whopping `425` components defined. When this is included in the project and included in the unit I am working in with `library`/`use` calls, I am experiencing 1-5s of delay when completing an item. During this delay, my entire editor freezes.

I am not sure if this is an issue specifically with `vhdl_ls`, Neovim's [`nvim-lspconfig`](https://github.com/neovim/nvim-lspconfig) (doubt), or the completion engine I am using, [`nvim-cmp-lsp`](https://github.com/hrsh7th/cmp-nvim-lsp) (potentially? [relevant reddit post](https://www.reddit.com/r/neovim/comments/x44f2l/nvim_lspconfig_nvimcmp_general_slowness_in_large/), but that says that it is probably the LSP server).

Obviously there are a couple ways to work around this issue
- Disable snippets for this LSP (as I have done for now). This isn't ideal, since the snippets are really helpful
- Remove the `unisim` library from the project. This is also not ideal, since then it can't lint these instantiations, and gives a ton of errors if you use them.

Since neither of these are _great_, would it be possible to add another work around by adding  the ability to disable snippets on a library by library basis? Something like
```toml
[libraries]
unisim.files = [
    '$XILINX_VIVADO/data/vhdl/src/unisims/unisim_VCOMP.vhd',
    '$XILINX_VIVADO/data/vhdl/src/unisims/primitive/*.vhd*',
]
unisim.is_third_party = true
unisim.disable_snippets = true
```

This might also be a symptom of another, deeper issue that can be solved. Since it would be _nice_ to have the instantiation snippets for the primitives that are defined here (although IMO not necessary. Most of the time I am not using these primitives, and if I am, I will probably copy the template out of Xilinx's docs).

Schottkyc137, 2024-08-29:
I just tried out the server with neovim and I also experience the lags. I will investigate further, but I don't see the same behaviour for VSCode, so this is potentially an issue of sorts with neovim.

Anselmo95, 2024-08-29:
Maybe it's not related but I noticed the lag too and after a bit of digging I found out that on my pc vhdl_ls is not killed after closing neovim.
 
So after two weeks I had around 200 instances of vhdl_ls running and killing all of them mostly solved the lag.

SethGower, 2024-08-30:
> Maybe it's not related but I noticed the lag too and after a bit of digging I found out that on my pc vhdl_ls is not killed after closing neovim.
> 
> So after two weeks I had around 200 instances of vhdl_ls running and killing all of them mostly solved the lag.

I have also noticed this before. Not sure if it's related. I just killed all orphaned processes, then tested, still getting the same lag.

SethGower, 2024-08-30:
> I just tried out the server with neovim and I also experience the lags. I will investigate further, but I don't see the same behaviour for VSCode, so this is potentially an issue of sorts with neovim.

yeah, I was talking with a coworker who uses VSCode, and he doesn't have the issue. I assumed this was a neovim specific problem, since VSCode is the "default" most people use. So if it were affecting that as well, this probably would have come up way earlier.

I tried a different completion engine, [`coq_nvim`](https://github.com/ms-jpq/coq_nvim), and that doesn't seem to have the same lag. So I am going to create a corresponding issue on [`nvim-cmp`](https://github.com/hrsh7th/nvim-cmp).

SethGower, 2024-08-30:
I _love_ that github showed the reference to my dotfiles repo commits....

pidgeon777, 2024-11-03:
I would like to kindly ask if this issue is still occuring.

SethGower, 2024-11-06:
I just updated all of my nvim plugins, and I don't think it is happening. However I am noticing a lot of lag with the LSP in general, I have noticed that before occasionally. I think this is causing a long delay in the snippets being completed, but it is no longer freezing my entire editor. So that's a plus I guess?

chrisbohens, 2025-03-13:
Hi, when I use nvim-cmp in a vhdl buffer that imports the unisim library, I experience delays of up to 3 seconds where Neovim freezes. When I remove the lsp as a source for nvim-cmp, the problem is gone. When I remove the unisim dependency, then the problem is gone as well. Interestingly in Helix I don't experience this problem at all.

pidgeon777, 2025-03-14:
> Hi, when I use nvim-cmp in a vhdl buffer that imports the unisim library, I experience delays of up to 3 seconds where Neovim freezes. When I remove the lsp as a source for nvim-cmp, the problem is [...]

Did you try using `blink.cmp`?

SethGower, 2025-03-18:
I was still having this issue with `nvim-cmp` last time I tried. Still using `coq_nvim` as I previously described. It would be nice if we can figure out the reason. However the issue I opened on `nvim_cmp` didn't seem to lead to anything.

@pidgeon777 I'll look into blink, thanks for the tip.

SethGower, 2025-03-18:
Quick testing shows that blink actually doesn't have the delay, and I really like blink compared to `coq` (I kinda hate coq...), so probably just gonna switch to blink completely.

pidgeon777, 2025-03-19:
> Quick testing shows that blink actually doesn't have the delay, and I really like blink compared to `coq` (I kinda hate coq...), so probably just gonna switch to blink completely. 

Good choice, you'll be happy with that 👍.