VHDL-LS/rust_hdl #342: Error in library

qarlosalberto, 2024-08-31:
I have 2 files:

```
-- half_adder.vhd

library ieee;
use ieee.std_logic_1164.all;

entity half_adder is 
  port (a, b : in std_logic;
        sum, carry : out std_logic
    );
end half_adder;

architecture arch of half_adder is
begin
  sum <= a xor b;
  carry <= a and b;
end arch;
```

```
library ieee;
use ieee.std_logic_1164.all;
use std.env.finish;

library mylib;

entity half_adder_process_tb is
end half_adder_process_tb;

architecture tb of half_adder_process_tb is
    signal a, b : std_logic;
    signal sum, carry : std_logic;
begin
    -- connecting testbench signals with half_adder.vhd
    UUT : entity mylib.half_adder port map (a => a, b => b, sum => sum, carry => carry);

    tb1 : process
        constant period: time := 20 ns;
        begin
            a <= '0';
            b <= '0';
            wait for period;
            assert ((sum = '0') and (carry = '0'))  -- expected output
            -- error will be reported if sum or carry is not 0
            report "test failed for input combination 00" severity failure;

            a <= '0';
            b <= '1';
            wait for period;
            assert ((sum = '1') and (carry = '0'))
            report "test failed for input combination 01" severity failure;

            a <= '1';
            b <= '0';
            wait for period;
            assert ((sum = '1') and (carry = '0'))
            report "test failed for input combination 10" severity failure;

            a <= '1';
            b <= '1';
            wait for period;
            assert ((sum = '0') and (carry = '1'))
            report "test failed for input combination 11" severity failure;

            finish;
        end process;
end tb;
```

An this `.vhdl_ls.toml` at home: 

```
[libraries]

work.files = [
  '/home/carlos/repo/vscode-terosHDL/packages/teroshdl/resources/project_manager/examples/ghdl/half_adder_tb.vhd',
]

mylib.files = [
  '/home/carlos/repo/vscode-terosHDL/packages/teroshdl/resources/project_manager/examples/ghdl/half_adder.vhd',
]
```

It was woking fine with old verions of `rust_hdl` but now I have to replace `work` by `default`. Is it expected?

Xcodo, 2024-08-31:
This is expected. See #226.

Perhaps this should be called out in the README, I can look at doing that.

qarlosalberto, 2024-08-31:
What should be the library name for files without lib?

closed 2024-08-31 by qarlosalberto

Xcodo, 2024-08-31:
Any name works (possibly except VHDL keywords). We use our product initials, but you could equally use `workLib` or something like that.