VHDL-LS/rust_hdl #345: Procedure ambiguos with string in TerosHDL

ArnePret, 2024-10-09:
I am using VHDL-LS through TerosHDL. The currently implemented version is 0.83.0, which seems to be the current version.
The attached screenshot shows the issue. Two procedures with different inputs are identified as ambiguos even though they are not.
Everything is identified correctly, when the to_string() function is used. A hard coded string does not work. When the string is passed as a constant everything is identified correctly as well (not in the screenshot).

[image]

Schottkyc137, 2024-10-09:
I'm not sure whether this is an error on the language server side. To me, this looks like the correct behaviour (although I may also have read the spec wrong). Did you try with a different tool to see whether that has the error too?
I'm unsure because, for example, for the following snippet:
```vhdl
library ieee;
    use ieee.std_logic_1164.all;

entity foo is
end foo;

architecture arch of foo is
	procedure f1(din: string) is
	begin
	end f1;

	procedure f1(din: std_logic_vector) is
	begin
	end f1;
begin
	process is
	begin
		f1("Hello");
	end process;
end arch;
```

`nvc` gives the error "type of string literal cannot be determined from the surrounding context" (which might arguably a better error message). 

In addition, in your example, it's somewhat apparent that you are trying to pass in a string, but consider the call
```
proc("00010", "1001");
```
Should the `std_logic_vector` or the `string` variant be called? The string literal is valid for both use cases.

Note that you can always disambiguate like so:
```
proc("Hello", string'("world"));
```

ArnePret, 2024-10-10:
I think you are correct and thank you for the clarification.

closed 2024-10-10 by ArnePret