VHDL-LS/rust_hdl #348: Unreachable code panic when instantiating a component with a generic type

maltaisn, 2024-10-21:
The following file:
```
library ieee;
use ieee.std_logic_1164.all;

entity Foo is
   port (
      clk : in std_logic);
end entity;

architecture Behavioral of Foo is

component Bar
   generic (
      type DATA_TYPE);
   port (
      clk : in std_logic);
end component;

begin

Inst_Bar : Bar
   generic map (
      DATA_TYPE => std_logic)
   port map (
      clk => clk);

end architecture;
```
Causes a panic:
```
thread '<unnamed>' panicked at vhdl_lang\src\named_entity\formal_region.rs:101:17:
internal error: entered unreachable code
```
I'm using v0.83.0.

jobtijhuis, 2024-10-30:
I was just about to report this problem, as I ran into it as well.

maltaisn, 2024-10-31:
I looked at the code and using subprograms or types in generic associations is simply not implemented. In fact generic associations are treated exactly like port associations at the moment.

On my side I just patched it by adding this line:

            AnyEntKind::Type(_) => TypeEnt::from_any(self.ent).unwrap(),

right after:

https://github.com/VHDL-LS/rust_hdl/blob/5cf0e4c6229534746fab1a0746a42f9a7e966b04/vhdl_lang/src/named_entity/formal_region.rs#L99

Which is definitely wrong but at least there's no panic.

Schottkyc137, 2024-11-04:
I just want to say that I'm on it, but I have decided against a quick fix because there currently is a lot of duplicate logic when analyzing interface lists (for example, the capabilities missing for entities is implemented for generic packages and generic subprograms) so I want to simplify the code a bit before rolling out the fix.

rpatel3001, 2024-12-11:
Just want to +1 for a slightly different but related issue. I'm getting the same panic with a package as generic:

```vhdl
package GenPkg is
generic (...);
end package GenPkg;

entity Foo is
end entity;

architecture Behavioral of Foo is

package GenPkgInst is new work.GenPkg generic map (...);

begin

Inst_Bar : entity work.Bar
   generic map (
      myGenPkg => GenPkgInst)
   port map (
      ...
   );

```

PR #382 (2025-05-09): Refactor map aspects
Closes #348, closes #379
closed 2025-05-09 by Schottkyc137