VHDL-LS/rust_hdl #349: lsp does not find any files in work library

Shoxx98, 2024-11-03:
[lsptest.zip](https://github.com/user-attachments/files/17611242/lsptest.zip)
Whenever I open a file, I get a warning that it does not belong to the project.
Whenever I use another file, (e.g. SubDesign) from top.vhd, i get the error in neovim: "no primary unit 'SubDesign' within library 'work'"

I am using NeoVim with Mason. The only thing I did to use rust_hdl is to have an entry in the ensure_installed list.

Shoxx98, 2024-11-03:
I mean using the work library specifically. If I name any library and use that one using the lsp, they just work. Since you cannot name the work library (because it shouldnt work like that), I am kind of lost here.

JHertz5, 2024-11-03:
No problem :)

JHertz5, 2024-11-04:
Hi @Shoxx98. If your problem has been resolved, please mark the issue as closed.

closed 2024-11-18 by Shoxx98