VHDL-LS/rust_hdl #351: Features like in Emacs VHDL mode planned?

plipK, 2024-11-22:
Are there any plans to implement more advanced features like the emacs vhdl mode in the future? I mean things like 
- custom models
- templates
- Port Copy/Past as (Entity/Component/Signals/Testbench/...)

I still use emacs for vhdl development, even though I hate the editor itself :-) But as long as there is no alternative tool that offers the same functionality for vhdl, I will not change.

jakobjung10, 2024-12-07:
@plipK 
The VSCode-extension [VHDL-by-HGB](https://github.com/HSD-ESD/VHDL-by-HGB) offers the Copy-Paste-Functionality from Emacs.

Schottkyc137, 2024-12-07:
+1. I think that the features you are looking would rather be found in the plugins for the various editors. VHDL-LS is only a language server and as such responsible for syntax checking, analysis, goto-declarations and some more, but generally not the features that you mention.

plipK, 2024-12-07:
Ok, apparently I wasn't really aware of the functions and limitations of the LSP. Thanks for the clarification!
And thanks also for the reference to VHDL-by-HGB, I'll have a look.

closed 2024-12-07 by plipK