VHDL-LS/rust_hdl #355: Library `work` not allowed anymore

Martinhinnerson, 2024-12-06:
Hi!
I work in a project where library 'work' is expected to be used globally for all files. After updating vhdl_ls it does not work anymore since the library 'work' is not allowed in the vhdl_ls.toml anymore.

I tried building vhdl_ls locally and just remove the check in `vhdl_lang/src/config.rs` that returns an error if `work` is used, and that seems to work.

What was the reason to disallow this?
If you cant remove it again, would it be possible to have a config option to allow library `work` again?

Xcodo, 2024-12-06:
Work can still be used in VHDL `use` statements, where it means "the current library". See this [blog post](https://www.sigasi.com/legacy/tech/work-not-vhdl-library/) for a brief explanation.

You don't need to change your VHDL files to use VHDL-LS, refering to modules as `use work.` is fine.

However, `work` is not allowed in the `vhdl_ls.toml` configuration file as the name of the library. Instead you can use any other name, for instance `work_lib` would be valid.

Martinhinnerson, 2024-12-06:
Thanks for the reply. I’m aware that work is not really a library, but I tried naming it defaultlib in the vhdl_ls.toml and it didn’t work. The language server did not recognize anything imported as work.

How does vhdl_ls know what is the ”current library”?

Schottkyc137, 2024-12-07:
> How does vhdl_ls know what is the ”current library”?

When you have a `vhdl_ls.toml` file,  you associate a library to each file:
```toml
[libraries]
libA.files = [
    "foo.vhd",
    "bar.vhd",
]

libB.files = [
    "baz.vhd",
    "foobar.,vhd",
]
```

When you edit `foo.vhd`, the `work` library will be set to `libA`. So visible symbols from the work library will include symbols from `foo.vhd` and `bar.vhd`. Similarly, if you edit stuff in `baz.vhd`, the `work` library is `libB` and therefore symbols from `baz.vhd` and `foobar.vhd`

DrPiWork, 2025-01-20:
Hi,
I second the initial request.
I have projects with source files located in several directories (which are common to several projects).
All files belong to the `work` library.
How can I specify the location of these directories ?

nathanaelhuffman, 2025-02-05:
> Hi, I second the initial request. I have projects with source files located in several directories (which are common to several projects). All files belong to the `work` library. How can I specify the location of these directories ?

I think that disallowing "work" as a name for a library is a perfectly reasonable design choice here, and one that helps prevent (or illuminate) common misunderstandings about the library, and I think that historically EDA tools have done a disservice by using "work" as an actual library name, confounding the issue.

Here's a partial example vhdl_ls.toml from one of my projects that I hope will aid in understanding:

```
[libraries.worklib]
files = [
    "hdl/ip/vhd/common/utils/time_pkg.vhd",
    "hdl/ip/vhd/axi_blocks/axil_common_pkg.vhd",
    "hdl/ip/vhd/axi_blocks/axilite_if_pkg.vhd",
    "hdl/ip/vhd/axi_blocks/axil_interconnect.vhd",
    "hdl/ip/vhd/espi/espi_target_top.vhd",
]

[libraries.vunit_lib]
files = [
    "vunit/vhdl/verification_components/src/sync_pkg.vhd",
    "unit/vhdl/verification_components/src/avalon_sink.vhd",
]
```
Here we have a bunch of different files, in different directories all compiled in to a library named "worklib" as far as vhdl-ls is concerned, and another library named "vunit_lib" with some files (which could also include separate directories etc)

This means that the `work` library of each of these files is the same, and it happens to be named worklib by the vhdl-ls tool. 

in espi_target_top.vhd, let's say I need access to the axil_common_pkg and axilite_if_pkg files.  Because they're all in the same library, each of them is in the `work` library, I can import them as follows:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.axil_common_pkg.all; -- I want axil_common_pkg from my same library, regardless of the library name
use work.axilite_if_pkg.all; -- same as above, give me the axilite_if_pkg from my current work library

entity espi_target_top is
...
```

Now, over in the vunit_lib library, say avalon_sink wants to include sync_pkg (from its current library)

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.sync_pkg.all; -- I want sync_pkg  from my current work library, regardless of the actual library name (vunit_lib in this case)

entity avalon_sink is
...
```

So `work` just means "the library into which I was compiled" and isn't actually a name proper.

Now for a final case, let's say we make some changes and espi_target_top now needs to pull in sync_pkg from vunit_lib:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.axil_common_pkg.all; -- I want axil_common_pkg from my same library, regardless of the library name
use work.axilite_if_pkg.all; -- same as above, give me the axilite_if_pkg from my current work library

library vunit_lib;  -- bring in another library that isn't my current library
use vunit_lib.sync_pkg.all; -- Now I want sync_pkg, but it's in another library than I am so I can't use work to reference it and have to use the library name

entity espi_target_top is
...
```

I hope those examples both help clarify how `work` is used in vhdl and provide some additional examples of using multiple libraries and putting files in different directories into the same library.

Martinhinnerson, 2025-02-06:
I would also like to withdraw my initial request.

I still am not sure what caused the problems I had but I think they where related to my neovim integration with vhdl_ls, rather than a library problem. In my case it’s working as expected now.

I’ll close the discussion and I’m happy with ’work’ not being allowed as a library name. Thanks for all the replies.

closed 2025-02-06 by Martinhinnerson