VHDL-LS/rust_hdl #358: [FEATURE] completion: nested records

jakobjung10, 2024-12-27:
Currently, members of nested records don't get LSP-completion-items.
The completion only works for members of the highest record in the hierarchy.
Nested completion would be a huge improvement! 

BR,
Jakob

Schottkyc137, 2024-12-28:
Just to confirm, the behaviour that is currently missing is the following:
```vhdl
type foo is record
    elem_a : bit;
    elem_b : bit;
end foo;

type bar is record
    elem_foo : foo;
end bar;

-- ...

signal x: bar;
signal y: bit;
y <= x.elem_foo. -- missing completions for `elem_a` and `elem_b`
```
correct?

jakobjung10, 2024-12-28:
Yes, this is correct :)

I also want to mention another missing feature.
Currently, record-completion is only provided for the right side of an assignment, but not for the left side.

```vhdl
type foo is record
    elem_a : bit;
    elem_b : bit;
end foo;

type bar is record
    elem_foo : foo;
end bar;

-- ...

signal x, next_x : bar;
next_x. -- missing completion for `elem_foo`    -- desired code: next_x.elem_foo.elem_a <= x.elem_foo.elem_b

```

Schottkyc137, 2024-12-29:
So the first feature request is pretty simple to implement and was just an oversight on my part.
The second one (completions for the left hand side of an assignment) is harder though. This is mainly because our current AST model doesn't deal too well with missing / extraneous tokens. Ideally (and also looking at stuff like better formatting, public API, more robust parsing e.t.c.), I would like to refactor the whole syntax parsing part. But that will likely take some time and thinking so I'm not sure when such a feature would be implemented.