VHDL-LS/rust_hdl #363: [Resources utilization]

ag943, 2025-01-29:
Hi,

I am using the extension on a linux shared server (in a vscode ssh host). I am noticing a very high cpu utilization from time to time (up to 4 cpus at 100%), and a mean utilization of about 2 cpus at 100%.

Is it an expected behavior ? 

Have a good day !

Schottkyc137, 2025-01-29:
That might be incorrect behavior. Can you provide some additional information to help narrow down the problem? Especially, the following metrics should be interesting:
- What version are you experiencing these issues? Did you also experience this problem with older versions?
- How did you install the program (I.e. using it via a VSCode / vim / other editor; compiling it from source, ...)
- how do you execute the program? (I.e. automatically using the extension, manual, ...)
- When exactly are you seeing the high resource utilization? Is it always, only when you modify the source code, during refactorings, ...)
- How big is the project you are analyzing, ideally in lines of code

ag943, 2025-01-29:
Thank you for your response. Here are the answers to your questions:

1. What version are you experiencing these issues? Did you also experience this problem with older versions?: **0.83.1**
2. How did you install the program (I.e. using it via a VSCode / vim / other editor extension; compiling it from source, ...): **Using the executable shipped with the vscode extension**
3. how do you execute the program? (I.e. automatically using the extension, manual execution, ...): **Automatically with vscode**
4. When exactly are you seeing the high resource utilization? Is it always, only when you modify the source code, during refactorings, ...) **I feel it is mainly when opening a session and then opening vhdl files, the utilization shoots up to several cores then slowly decreases to a normal level (10%)**
5. How big is the project you are analyzing, ideally in lines of code: **The projects sizes are around 800 000 lines of vhdl**

Schottkyc137, 2025-02-01:
This could be intended behavior. On startup, the server parses and analyzes as many files in parallel as possible, potentially causing the high resource utilization.
I'm not sure whether the VSCode extension allows a user to limit the number of threads. If the high utilization bothers you, and that option doesn't exist in the extension, I think we could add a flag to the extension to limit the number of threads.

ag943, 2025-02-04:
For my use case it would be great to have a way to limit the amount of threads, as high CPU load can have an impact on the work of my colleagues (as resources are shared). Thank you for the consideration !