VHDL-LS/rust_hdl #364: [Feature Request] Warning on null ranges

JHertz5, 2025-02-06:
I would like to request a feature. I was recently helping a junior engineer to debug a mysterious error that he was struggling with. After searching high and low, we discovered that they had used `to` instead of `downto` when defining a range, resulting in a null range, which was only caused a cryptic error message from the simulation tool several layers down in the hierarchy.

I understand that use of null ranges is valid VHDL and I understand if you don't want to create warnings on valid code, but I think that it would helpful to have a warning shown in cases like this, perhaps one that can be suppressed by the user (or even, is suppressed by default and can be un-suppressed by the user) for people who want to use null ranges?

Specifically, I would like to see warnings on `X to Y` and `Y downto X`, where X < Y in both cases, e.g.

```vhdl
library IEEE;
use IEEE.std_logic_1164.all;

package test is

  subtype null_range_to     is integer range 1 to 0;
  subtype null_range_downto is integer range 0 downto 1;

  subtype slv_null_range_to     is std_logic_vector(1 to 0);
  subtype slv_null_range_downto is std_logic_vector(0 downto 1);

end package test;
```

oscargus, 2025-03-12:
(Slightly related: it seems like some tools warns for this. https://github.com/VUnit/vunit/pull/1102)

nselvara, 2025-04-22:
I think it should be warned as well. And if one wants to remove the warning could use this feature/PR: https://github.com/VHDL-LS/rust_hdl/pull/372