VHDL-LS/rust_hdl #369: Ignore any references to vunit_lib (do not report errors)

obruendl, 2025-03-11:
I asked below question on gitter but I did not get a reply by now - so I try it again though a GitHub issue:

## Initial Question

I am just starting to use VHDL-LS. Is there any documentation of the options I can set in the TOML?

Specifically I have the following question:

I use VUnit (and it is installed locally - so in a machine dependent location - through PIP). I would want to avoid VHDL-LS reporting errors on use-clauses like the one below. Can I somehow configure VHDL-LS to not report that it does not find the sources for anything in vunit_lib?
library vunit_lib;
context vunit_lib.vunit_context;
context vunit_[lib.com](http://lib.com/)_context;
context vunit_[lib.vc](http://lib.vc/)_context;

I already tried "vunit_[lib.is](http://lib.is/)_third_party = true" but this line alone didn't do the job.

Can I somehow (e.g. by a comment in my VHDL) ignore a line? Sometimes I have unused generics for certain reasons (e.g. to not break reverse compatibility) and I'd like to avoid VHDL-LS reporting them as problems constantly.

## First Reply (not resolving the issue)

Please look at the example in the Configurations section of the README.
https://github.com/VHDL-LS/rust_hdl?tab=readme-ov-file#configuration
It has an example of adding the Unisim library in the TOML file.

### Follow-up from my side (no answer received)

I looked at this example. That's how arrived to the "vunit_[lib.is](http://lib.is/)_third__party = true" idea, which did not work.

The point is that I cannot implement the line below, because the install location of vunit_lib is machine dependent (because the vunit_lib is installed through PIP):

UNISIM.files = [
'C:\Xilinx\Vivado\2023.1\data\vhdl\src\unisims\unisim_VCOMP.vhd',
]

If you want to reproduce: The code (with my current TOML) is here:
https://github.com/open-logic/open-logic/tree/feature/VHDL-LS

obruendl, 2025-03-11:
To be honest, I would prefer just ignoring anything in _vunit_lib_ over generating the TOML from VUnit.

Technically the solution makes sense. But the more dependencies between different tools (editor, simulator, etc.) is added, the more complex the project gets - which can be daunting to potential contributors and finally lead to them not contributing. So I  try to keep things as simple as possible.

Bottomline: Is there a way to ignore any errors towards _vunit_lib_ (e.g. through the TOML or through some VHDL comments instructing VHDL-LS to ignore certain lines in the file)?

If not - I might use the "generate TOML from VUnit" option as workaround. But for me it would clearly be preferrable to have a standalone TOML that does not require VUnit execution.

LukasVik, 2025-03-12:
Even if the unresolved library/use statements could be ignored, all VUnit types/functions/entities used in your testbench will also generate errors. There is no way for the language server to know if these missing things are from VUnit (which should be ignored), or if they are missing due to a typo/error/mistake (which should be reported). I don't know how it would be theoretically possible to ignore all errors that stem from VUnit usage.

I don't think there is an equivalent function in any other language server. When using Python/C++/Rust in VSCode I have to add my libraries manually, or there will be warnings. I think that's just how it is.

In the case of vhdl_ls I think generating the configuration file is the best approach: https://github.com/tsfpga/tsfpga/blob/main/tsfpga/create_vhdl_ls_config.py
I call this function from my simulation script, which is called regularly when doing development, so that my config is always up to date.

obruendl, 2025-05-24:
@LukasVik 

OK, works perfectly fine for me now.

I now have a licensing question - Would you be okay if I copy this SINGLE FILE to open-logic and I publish it under PSI HDL License? The file is under BSD-3 and I'd like to not overcomplicate the licensing of Open Logic by having to state that there are certain files under other although permissive licenses.

Similar question, may I set the copyright to myself for the edited version? For new content in open-logic I have the policy to not have other copyright holders so I potentially could change the license in future.

It's really only about this single create_vhdl_ls_config.py file only.

obruendl, 2025-05-24:
@LukasVik 

That's what I assumed ;-) But I tend to stay clean on copyrights and the like. Thanks for the quick OK - that allows me to proceed with the next release (which is a nice one, you'll see).