VHDL-LS/rust_hdl #373: How to format code using vhdl_ls

jay101-STAR, 2025-03-21:
I don’t know how to use vhdl_ls to format code,i install vhdl_ls in lazyvim and I read the README documentation carefully, but I’m still unclear about how to use it. Perhaps vhdl_ls doesn’t support code formatting, but I found some issues related to formatting code, which makes me think it might be possible. I don’t want to bother you, but I’ve spent over three hours trying to find out how to use vhdl_ls for code formatting.

Schottkyc137, 2025-03-22:
While `vhdl_ls` has a code formatter, it is nowhere near the state that one could use it in a productive environment, so this feature is currently experimental only. Furthermore, I will not continue development on this until https://github.com/Schottkyc137/rust_hdl/pull/1 is fully implemented, since the current AST model isn't too well suited for creating a formatter.

jay101-STAR, 2025-03-23:
ok,Thank you for your response

closed 2025-03-23 by jay101-STAR