VHDL-LS/rust_hdl #378: [BUG] False warning for generic function used in clock edge detection

nselvara, 2025-04-25:
**Bug description**:

Hi there, another day of bug reporting. 😬
I'm getting the following warning from VHDL LS regarding an incomplete sensitivity list when using a **generic function** (e.g., `rising_edge` or `falling_edge`) in the entity’s generic declaration: 

**The signal 'internal_signal' is not read in the sensitivity list `vhdl ls(missing_in_sensitivity_list)`**

This occurs when the clock edge detection function is passed as a generic, based on user configuration. While I understand why this is flagged - the analyser likely searches for a direct call to `rising_edge()` or `falling_edge()` as a pattern (similar to what synthesis tools do) - this is technically correct VHDL.

> [!NOTE]
> Attempting to suppress the warning using the `-- vhdl_ls off` pragma on the line causes other errors to be reported, so it isn't a viable workaround.

---

**Minimal reproducible example (MRE):**

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity generic_clk_edge_function_warning is
    generic (
        -- Function in the generic, which triggers the warning
        function active_edge(signal s: std_ulogic) return boolean;
        DATA_WIDTH: positive := 8
    );
    port (
        clk: in std_ulogic; -- Dummy input clock
        data_out: out std_ulogic -- Dummy output
    );
end entity;

architecture behavioural of generic_clk_edge_function_warning is
    signal internal_signal: std_ulogic := '0';
begin
    process (clk)
    -- ^ Warning reported here: missing_in_sensitivity_list
    begin
        if active_edge(clk) then  -- Trigger: function call via generic
            internal_signal <= not internal_signal;
        end if;
    end process;

    data_out <= internal_signal;
end architecture;
```

Schottkyc137, 2025-04-26:
Closed as part of [6def6bc](https://github.com/VHDL-LS/rust_hdl/commit/6def6bcd25033442809ae99937507e4b19c3ea27)

closed 2025-04-26 by Schottkyc137

nselvara, 2025-04-28:
Damn that was quick, thx yaa for your your service 🚀