VHDL-LS/rust_hdl #388: [Feature Request] Nightly build pipeline and preview version of VSCode extension

nselvara, 2025-05-15:
Hi there,

Thank you very much again for the valuable contributions.
I'd like to propose adding support for **nightly builds** of the language server as well as a **preview release** channel for the VSCode extension.

This would allow contributors and advanced users to:
- Test upcoming features and bug fixes early
- Provide feedback before stable releases
- Validate fixes for specific issues (e.g. linked to `main` branch commits)

---

**Proposed features:**
- A GitHub Actions-based nightly CI build for the VHDL LS server binary (e.g., uploaded as an artifact or GitHub release with `nightly` tag)
- A `preview` version of the VSCode extension (via separate release channel or prerelease tag on the marketplace)

---

**Why it’s useful:**
- Enables faster testing of bug fixes and experimental features
- Reduces feedback loops for contributors
- Helps in catching regressions early
- Encourages community engagement without destabilising the main release

---

**Optional ideas:**
- Include the git commit SHA and build timestamp in the LSP version string
- Provide instructions in the README on how to switch to preview/nightly builds

Thanks a lot for your great work - I'd be happy to help test things if this gets set up!

Xcodo, 2025-05-15:
I could potentially work on this (perhaps slowly!) if @Schottkyc137 is in favour of it.

Xcodo, 2025-05-15:
I noticed an exiting PR, VHDL-LS/rust_hdl_vscode#89, that could be an alternative way of doing this too. It has support for "latest" or a defined version, but could also have support for "nightly".

nselvara, 2025-05-15:
Yeah, I think this is easily addable and the PR is anyway open - so the change can be added there as well.

Schottkyc137, 2025-05-15:
I'm generally very much in favour of this, but don't have time to implement it / it would be low on my priority list. So it's nice to see that there are people interested in developing this. Therefore, It's a +1 from my side.
Concerning the aforementioned PR, I have sort of stopped development for two reasons. The first obviously is time, but the second is the lack of test in `rust_hdl_vscode`. Essentially I was worried that, if I submit these changes, more things would break than the things that would work. Additionally, there are other popular solutions that already integrate `vhdl_ls`, so the native extension also landed low on my priority list. But maybe it's worth putting some more effort into this.