VHDL-LS/rust_hdl #393: Completion Type Case Config

n8tlarsen, 2025-06-06:
Just recently updated to version 0.85 from a much older version of vhdl-ls and noticed additional completion options shown as `TypeParameter`s in Neovim. Since VHDL is case-insensitive I typically write entire files using lowercase characters. Is there a way to have the auto completion for `TypeParameter` and other similar completions to be written in lowercase instead of all capitals?

Schottkyc137, 2025-06-08:
I think this should be easy enough to implement. The only thing I'm not sure is where one would specify the casing of identifiers (i.e. whether one would use lowercase, all uppercase, first letter capitalized, e.t.c.).
One natural way is the `vhdl_ls.toml` file but this might not be the ideal place for stuff like configurations since one should be able to easily auto-generate this file. Therefore, I would prefer some auxiliary file for stuff like linting. But then again this would mean that there are two files for the language server. What is your opinion?

n8tlarsen, 2025-06-08:
I've considered auto-generation of the `vhdl_ls.toml` file so I get where you're coming from, but it should be easy enough for a tool to ignore or generate its own `[config]` table. Plus the `[lint]` table already exists within the `vhdl_ls.toml` so I would probably prefer just one file to control all things VHDL-LS.

Not sure how difficult this would be or how editors interact with an LSP server, but this could alternatively be something specified at launch time or runtime with a command passed to the server. At least with Neovim plugins (I know VS Code also has plugin options), there are often configurations and settings that can be passed to the plugin as it launches or after launch to set various configurations.

AndrzejKowalski9917, 2025-06-12:
I like the idea from @n8tlarsen and would also support the configuration via the already existing toml file.

Schottkyc137, 2025-08-16:
This is now possible. However, I think that for productive use there have to be some more elaborate configuration options. I’ve [started a discussion on GitHub](https://github.com/VHDL-LS/rust_hdl/discussions/400) and would love to hear your thoughts on it. Any input would be greatly appreciated!

PR #399 (2025-08-16): Add support for customizable case transformation in completions
Closes #393