VHDL-LS/rust_hdl #402: "Found no match for operator" error on logical operations on arrays

thoste, 2025-08-29:
Hi, 
I've found an issue where you get the "Found no match for operator" error if you have logical operations on arrays.
This sould be valid code in VHDL. 

[image]

Code example:
```VHDL
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity foo is

end entity foo;

architecture rtl of foo is
   type t_bool_array is array (natural range <>) of boolean;

   signal result  : t_bool_array(3 downto 0);
   signal a_array : t_bool_array(3 downto 0);
   signal b_array : t_bool_array(3 downto 0);
begin

   result <= a_array or b_array;
   result <= a_array and b_array;

end architecture;
```

riverolfe, 2026-09-30:
The code is valid VHDL. IEEE 1076-1993, 7.2.1 (Logical operators):

> The logical operators and, or, nand, nor, xor, xnor, and not are defined for predefined types BIT and BOOLEAN. They are also defined for any one-dimensional array type whose element type is BIT or BOOLEAN.

These operators are implicitly declared with the type; the tools don't add them. `boolean_vector` gets its operators the same way: `std.standard` declares it as `array (natural range <>) of boolean` with no explicit `"or"` (see `ghdl --disp-standard --std=08`). GHDL accepts the example with plain `-a --std=93` and `--std=08`.

Still reproduces in vhdl_ls 0.88.0, also for `not` and for user-defined arrays of `bit`.

Schottkyc137, 2026-10-04:
Thanks to all for reporting & the LRM digging. This has been fixed with e6a7d7261ea6f1c97a7e183bc3249eab4d422ec9

closed 2026-10-04 by Schottkyc137